thonny/thonny #2503: Request: Improved Variable Explorer

mu-us61, 2022-10-13:
Hello it would be perfect if the variable explorer is detailed like "Spyder IDE" , especially for beginners , seeing types would be very helpful

[image]

aivarannamaa, 2022-10-18:
Thank you for the suggestion!

I understand the benefit of having explicit type column in the Variables view, but it would make the initial impression of the UI more complex. Luckily, for simple values the type is easy to infer from the value's repr string.

Even if I added this column, I wouldn't put it before the value, before this could give the wrong impression, that type belongs to the variable (but this is not the case in Python). In my opinion, the principle, that a value has type, is nicely communicated by Thonny's Object inspector (View => Object inspector) -- it always shows a value (together with its type), even if the value is approached via a variable.

mu-us61, 2022-10-18:
Ok thanks, I missed the Object inspector, that is very nice and far more detailed

closed 2022-10-18 by mu-us61