thonny/thonny #2505: Packages aren't visible in package manager after installing (Thonny 4.0.1, Windows 64 bit)

Pet71, 2022-10-14:
Tried to install some packages for Raspberry Pico, but after installation I can't see installed package in the list:

[image]

I didn't get error any message during the installation process, packages simply don't appear and of course I can't use them. I used these libraries with earlier versions without any issue.
Tried to:
- searching for this error on the net, but couldn't find any solution.
- run Thonny as administrator
- run 32 bit/64 bit portable versions
- reinstall Thonny several times
... But no success.
Thank you for support in advance!

aivarannamaa, 2022-10-18:
Thank you for the report!

When the installation dialog (with progress bar) appears, please click the blue link in the bottom of the dialog. This will open the log. It could be helpful if you copied the log text here.

Pet71, 2022-10-21:
Thank you for reply.
When I try to install package from PyPI, the installation window closes immediately.
If I try to install package from file, got this empty box, when clicking on the link:

[image]

aivarannamaa, 2022-10-22:
I may find more information from the log files. I'd be glad if you performed following process:

* Select "Tools => Options => General" and check "Debug mode" checkbox
* Restart Thonny
* Select "Tools => Open Thonny data folder" and keep the directory open
* Try to reproduce the issue (open package manager, try to install a package)
* Close Thonny
* Send me frontend.log and backend.log from the data directory.

Pet71, 2022-10-27:
Ok, there are the .log files:

[backend.log](https://github.com/thonny/thonny/files/9881895/backend.log)
[frontend.log](https://github.com/thonny/thonny/files/9881896/frontend.log)

aivarannamaa, 2022-10-29:
Thanks! The back-end log gives me the exception. I'll try to reproduce and fix this.

aivarannamaa, 2023-01-16:
https://github.com/thonny/thonny/releases/tag/v4.0.2 is now out. Please reopen if you still have the same problem.