thonny/thonny #2507: Thonny doesnt like Compiz

Skilowi, 2022-10-17:
Thonny 4.0.1, installed today by pip3

Any way how to prevent those glitches from appearing? Not using Compiz isn't acceptable answer.

Video of issue + Photo of neofetch, might be useful 

https://streamable.com/rp99dg

[image]

aivarannamaa, 2022-10-18:
Thank you for the report!

Do Python's official editor -- IDLE -- run fine in your system? I'm asking, because it uses the same UI toolkit (Tk).

Skilowi, 2022-10-18:
IDLE has also that glitch, but those beams of light are darker, and they appear rarely

aivarannamaa, 2022-10-19:
I'm afraid this is the problem in Tk and therefore out of my reach. If you can reproduce this in a plain Tcl/Tk application (eg. [wish](https://packages.ubuntu.com/impish/wish)), then it makes sense to raise the issue in Tcl/Tk mailing list or bug tracker. Otherwise you could notify IDLE developers about the glitch (https://github.com/python/cpython/issues) and they may come up with a better answer than I can.

closed 2023-01-28 by aivarannamaa