thonny/thonny #2509: Another OSError: [WinError 87] The parameter is incorrect

Sherman-c, 2022-10-18:
Hi, Hope someone could help me
I tried to install umqtt.simple via Manage Package, it returned 
OSError: [WinError 87] The parameter is incorrect

when I tried to install it manually , i found the following trackback message

C:\Program Files (x86)\Thonny>python -m pipkin  install micropython-umqtt.simple 1.3.4
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files (x86)\Thonny\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\__main__.py", line 62, in <module>
    sys.exit(main(sys.argv[1:]))
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\__init__.py", line 48, in main
    command_handler(**args_dict)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\session.py", line 100, in install
    self._populate_venv()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\session.py", line 550, in _populate_venv
    self._clear_venv()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\session.py", line 528, in _clear_venv
    sp_path = self._get_venv_site_packages_path()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\session.py", line 525, in _get_venv_site_packages_path
    return get_venv_site_packages_path(self._venv_dir)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\pipkin\util.py", line 67, in get_venv_site_packages_path
    result = subprocess.check_output(
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files (x86)\Thonny\lib\subprocess.py", line 1438, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 87] The parameter is incorrect

The same error found on both Thonny 4.0.0 and 4.0.1, and not limited on umqtt.simple, but other modules as well

Could anyone able to tell me what I ran wrong

Regards
Sherman

aivarannamaa, 2022-10-18:
Thank you for the report!

The error message is similar to #1000, but the context is different.

Are you able to run simple programs with F5?

Sherman-c, 2022-10-18:
Hi Aivar,
Thank you for your fast response.  I'm not sure I understand your question
well,  I have no issue running any other program with thonny as long as
those required modules/libraries were already installed, like machine,
network and time.

BTW What is F5?

Best regards
Sherman

aivarannamaa, 2022-10-19:
F5 is the keyboard shortcut for running the script in Thonny.

It's good to know that running program works for you. Unfortunately this `[WinError 87]` is a tough one -- I haven't been able to reproduce it so far and I don't know what may cause it.

I'll try a tweak in pipkin and let you know when it is ready to test.

Sherman-c, 2022-10-19:
Hi Aivar

I have done some debugging on suprocess.py, the error was raised by
               hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                                         # no special security
                                         None, None,
                                         int(not close_fds),
                                         creationflags,
                                         env,
                                         cwd,
                                         startupinfo)

while
executable is None
args
= C:\Users\<username>\AppData\Local\pipkin\cache\workspaces\1c7bfecf12769f9897e393af77815ec0\Scripts\python.exe
-c "import site; print([p for p in site.getsitepackages() if
'site-packages' in p][0])"
env is None
cwd is None

the executable was passed by class Popen  __init__()
which called by line 501
with Popen(*popenargs, **kwargs) as process:

popenargs =
['C:\\Users\\<username>\\AppData\\Local\\pipkin\\cache\\workspaces\\1c7bfecf12769f9897e393af77815ec0\\Scripts\\python.exe',
'-c', "import site; print([p for p in site.getsitepackages() if
'site-packages' in p][0])"]
kwargs = {'stdout': -1, 'text': True, 'stdin': -3}

I tried to pass the values on executable, args, and env, but it returned
another error
AttributeError: 'str' object has no attribute 'keys'

Hope this information is useful.

Regards
Sherman

aivarannamaa, 2022-10-19:
Thanks!

Does C:\Users\<username>\AppData\Local\pipkin\cache\workspaces\1c7bfecf12769f9897e393af77815ec0\Scripts\python.exe exist on your machine?

Sherman-c, 2022-10-19:
Hi Aivar

Yes, it does, the version of the python.exe is 3.10.6

Best regards,
Sherman

Sherman-c, 2022-10-20:
Hi Aivar,

BTW if I executed the line found args,

C:\Users\<username>\AppData\Local\pipkin\cache\workspaces\1c7bfecf12769f9897e393af77815ec0\Scripts\python.exe
-c "import site; print([p for p in site.getsitepackages() if
'site-packages' in p][0])"

on cmd prompt, it returned

C:\Users\<username>\AppData\Local\pipkin\cache\workspaces\1c7bfecf12769f9897e393af77815ec0\lib\site-packages

without any error, althought the folder  exists,
C:\Users\<username>\AppData\Local\pipkin\cache\workspaces\1c7bfecf12769f9897e393af77815ec0\lib\site-packages
it is empty, nothing there

Regards
Sherman

aivarannamaa, 2022-10-20:
It looks like in most cases people relate Error 87 to some extra long path or extra large environment variable.

Please open the terminal (eg. cmd.exe or Powershell) just like you did when you executed pipkin directly and received the error. When you look your environment variables with `set`, do you notice an extra long variable value? Can you tell me the number of characters in the whole output of the `set` command?

Which Windows version are you using?

Sherman-c, 2022-10-21:
Hi Aivar,

Thanks for the follow up, SET returns total 2030 characters in 40 lines,
the longer variable PATH contains 610 characters, registry variable
LongPathsEnabled is set to 1 under
HKLM\SYSYEM\CurrentControlSet\Control\FileSystem
The window version is
Window 10 Pro 21H2 with Window Feature Experience Pack 120.2212.4180.0

BTW I'm not sure it helps, I have platformIO installed under VsCode,
although I could not run platformIO with VsCode GUI but CLI only, the
python version 3.9.13

Best Regards
Sherman

aivarannamaa, 2022-10-22:
I now uploaded a new version of pipkin (1.0b5) to PyPI, which uses a slightly different approach for starting a sub-process. Please try, whether this fixes your issue.

Sherman-c, 2022-10-23:
Hi Aivar

I got a new error after upgrade pipkin to 1.0b5, I received the following
error message under Tool|Manager package

C:\Users\<username>\AppData\Local\pipkin\cache\workspaces\1c7bfecf12769f9897e393af77815ec0\Scripts\python.exe:
No module named pip
Error Command
'['C:\\Users\\<username>\\AppData\\Local\\pipkin\\cache\\workspaces\\1c7bfecf12769f9897e393af77815ec0\\Scripts\\python.exe',
'-I', '-m', 'pip', '--no-color', '--disable-pip-version-check',
'--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517',
'--upgrade-strategy', 'only-if-needed', 'micropython-umqtt.simple',
'--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.

if I ran it with CMD under Thonny, it returned the following
>python -I -m pip --no-color --disable-pip-version-check --trusted-host
127.0.0.1 install --no-compile --use-pep517 --upgrade-strategy
only-if-needed micropython-umqtt.simple --index-url http://127.0.0.1:36628
Looking in indexes: http://127.0.0.1:36628
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object
at 0x0000016D28C4C160>: Failed to establish a new connection: [WinError
10061] No connection could be made because the target machine actively
refused it')': /micropython-umqtt-simple/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object
at 0x0000016D28C23400>: Failed to establish a new connection: [WinError
10061] No connection could be made because the target machine actively
refused it')': /micropython-umqtt-simple/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object
at 0x0000016D28C4C5B0>: Failed to establish a new connection: [WinError
10061] No connection could be made because the target machine actively
refused it')': /micropython-umqtt-simple/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object
at 0x0000016D28C4C7C0>: Failed to establish a new connection: [WinError
10061] No connection could be made because the target machine actively
refused it')': /micropython-umqtt-simple/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None,
status=None)) after connection broken by
'NewConnectionError('<pip._vendor.urllib3.connection.HTTPConnection object
at 0x0000016D28C4C9D0>: Failed to establish a new connection: [WinError
10061] No connection could be made because the target machine actively
refused it')': /micropython-umqtt-simple/
ERROR: Could not find a version that satisfies the requirement
micropython-umqtt.simple (from versions: none)
ERROR: No matching distribution found for micropython-umqtt.simple

Regards
Sherman

aivarannamaa, 2022-10-29:
Thanks for the info!

This seems to be the same case that was reported here: https://github.com/aivarannamaa/pipkin/issues/2
I'll try to reproduce this later.

aivarannamaa, 2023-01-16:
https://github.com/thonny/thonny/releases/tag/v4.0.2 is now out. Please reopen if you still have the same problem.

closed 2023-01-16 by aivarannamaa

Sherman-c, 2023-01-16:
Hi Aivar,

Thanks for the follow up.

Regards
Sherman

Sherman-c, 2023-01-17:
Hi Aivar

With Thonny 4.0.2, i got the following errors, on the 1st installation,

install micropython-umqtt.simple
Preparing working environment ...
Collecting pip==22.2.2
  Downloading pip-22.2.2-py3-none-any.whl (2.0 MB)
     ---------------------------------------- 2.0/2.0 MB 5.7 MB/s eta
0:00:00
Collecting setuptools==65.4.1
  Downloading setuptools-65.4.1-py3-none-any.whl (1.2 MB)
     ---------------------------------------- 1.2/1.2 MB 7.1 MB/s eta
0:00:00
Collecting wheel==0.38.4
  Downloading wheel-0.38.4-py3-none-any.whl (36 kB)
Installing collected packages: wheel, setuptools, pip
  Attempting uninstall: setuptools
    Found existing installation: setuptools 65.5.0
    Uninstalling setuptools-65.5.0:
      Successfully uninstalled setuptools-65.5.0
  Attempting uninstall: pip
    Found existing installation: pip 22.3.1
    Uninstalling pip-22.3.1:
      Successfully uninstalled pip-22.3.1
Successfully installed pip-22.2.2 setuptools-65.4.1 wheel-0.38.4
Error [WinError 87] The parameter is incorrect

however when i re-installed the same package again, it ran through ok, the
package successfully installed.  I guest the problem indeed fixed.  Thanks.

Happy New Year.
Best mregards
Sherman