thonny/thonny #2512: thonny is not opening : ImportError DLL load failed while importing_ctypes: The parameteri incorrect

CLASHERNSR, 2022-10-19:
__Traceback (most recent call last)

File <string, fine 3, in <module> File "CAUsers\Sai dj\AppData\Local\Programs\Thonny\lib\runpy.py line 203, in run module mod name, mod spec, codes get module details(mod_name)

File "C\Users\Sai d\AppData\Local\Programs\Thonny\lib\runpy.py

line 144, in get module details

return get module details(pkg_main name error

File "CAUsers\Sai dAppData\Local\Programs Thonnylibrunpy.py

line 111, in get module details

import (pkg name)

File "C:\Users\Sai

dj\AppData\Local\Programs\Thonnylib\site-packages\thonny\_init_ py", line 155, in <module> THONNY USER DIR=compute thonny user dir File "C:\Users\Sai dj\AppData\Local Programs Thonny\lib\site-packages\thonnyinit

py, line 77, in compute thonny user dir

elif is portable

File CAUsers\Sai

dAppData\Local Programs Thonny\lib\site-packages\thonny_init_ py, line 133, in is portable from ctypes import windil File "C:\Users\Sa

dj\AppData\Local Programs Thonny\lib\ctypes\_init_.py", line 7, in

<module> from ctypes import Union, Structure, Array

ImportError DLL load failed while importing_ctypes: The parameteri incorrect

_the thing is i am kinda new in this and i don't even know if its a basic error or not
So please just help me out and tell me how to solve this Error ASAP_

aivarannamaa, 2022-10-22:
Thank you for the report!

Which Windows version are you using? Which Thonny version?

navisr, 2022-10-22:
> Thank you for the report!
> 
> Which Windows version are you using? Which Thonny version?

Thonny installer with 32-bit python 3.8 (recommended) on windows 7 32-bit!

navisr, 2022-10-23:
Try to install an older version of thonny and rewrite, and it will work!

closed 2022-10-23 by navisr