thonny/thonny #2513: PROBLEM IN THONNY'S BACK-END:

Darklight29, 2022-10-21:
I'm using a Windows computer to to load a Pico W. I can see the Pico when I load rp2-pico-w-20221011-unstable-v1.19.1-529-g4fc543c82.uf2. My research says that this is the file to use for the program.

After typing in the program and trying to run it, I get the following msg, in a loop:

PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.

Is this hardware or software?

if hardware, I'll contact the company and ask for a replacement, if not, would you point me in the direction for the resources I can use to hopefully correct the problem. I've already passed this info to the program creator, no answer as of yet.

Thank you.
Mike

aivarannamaa, 2022-10-22:
Thank you for the report!

I haven't seen this kind of error before. I suggest to first try to connect to your Pico via a simpler program, eg Putty (consult e.g. https://picockpit.com/raspberry-pi/raspberry-pi-pico-and-micropython-on-windows/#PuTTY for more details). Make sure to close Thonny first.

If you can successfully connect with Putty and still can't connect with Thonny, then I need to investigate this further. Otherwise it's wisest to turn to https://forums.raspberrypi.com/viewforum.php?f=143

marcelomaiajuvencio, 2022-11-29:
Hello!!
Look
https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory

Resetting Flash memory

aivarannamaa, 2022-12-28:
@Darklight29, have you been able to connect with Putty?

Darklight29, 2022-12-28:
No

I am very frustrated!

I hope things get better, but I am glad when they don't get worse

aivarannamaa, 2022-12-29:
@Darklight29, if you're not able to connect with Putty, then your issue is not specific to Thonny. I recommend asking for advice at https://forums.raspberrypi.com/viewforum.php?f=146

closed 2022-12-29 by aivarannamaa

kmieciu2003, 2025-05-07:
I know I'm bringing up an old topic, but maybe it will help someone. I have a humidity, pressure, and temperature sensor on a Pico W:
[image]
With its own power source:
[image]
I use it in the greenhouse during the summer season.
The problem is that I need to change the network name because I moved the device to a different location, and after connecting it to the computer, I get the same error as the colleague. Putty communicates correctly:
[image]
The USB mass storage mode also works, and I can easily update the MicroPython version via Thonny.
Some suggested that the issue was a voltage drop in the power supply when peripherals are connected (in my case, a display and a sensor), but I tried connecting it with a long cable, a short one, directly to the computer, through a USB hub with its own power supply, and finally with batteries connected directly to the Pico. No result.
Eventually, I realized that I had encountered a similar problem before, when a bare board worked, but stopped working after connecting peripherals via I2C.
And now here’s the key point — after disconnecting the display from pin 36 (3V3 OUT)
[image]
, voila:
[image]
So it really looks like a power supply issue, but not on the computer-to-board line — rather internally on the board itself.