thonny/thonny #2515: Thonny 4.0.1 does not save/run with more than one instance

BNNorman, 2022-10-22:
I have installed the downloaded EXE for all users.
I am running on windows 10 and I have unchecked "Allow only single Thonny instance". 
I am trying to test a simple program to check Uart pico to pico comms and would like to see the messages received by each pico.
I can open two instances of Thonny and load the test program from each but when I click Run I get the following message. Note: The same applies when trying to save yet the instances know which COM device they are working with when I open a file.

[image]

I used revo to completely uninstall before re-installing Thonny but it made no difference.
I might try pip install instead.

BNNorman, 2022-10-22:
Same problem with the pip installed thonny ...
```
C:\WINDOWS\system32>python -m thonny
12:45:40.883 INFO    thonny: Thonny version: 4.0.1
12:45:40.883 INFO    thonny: cwd: C:\WINDOWS\system32
12:45:40.884 INFO    thonny: original argv: ['python', '-m', 'thonny']
12:45:40.884 INFO    thonny: sys.executable: C:\Users\brian.000\AppData\Local\Programs\Python\Python310\python.exe
12:45:40.884 INFO    thonny: sys.argv: ['C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\thonny\\__main__.py']
12:45:40.884 INFO    thonny: sys.path: ['C:\\WINDOWS\\system32', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\python310.zip', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\DLLs', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\win32', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\win32\\lib', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\Pythonwin']
12:45:40.885 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
12:45:43.626 INFO    thonny.running: Starting the backend: ['C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\python.exe', '-u', '-B', 'C:\\Users\\brian.000\\AppData\\Local\\Programs\\Python\\Python310\\lib\\site-packages\\thonny\\plugins\\circuitpython\\cirpy_back.py', "{'clean': False, 'port': 'COM7', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"] C:\Users\brian.000
12:45:44.671 INFO    thonny.plugins.micropython.mp_front: Falling back to serial.tools.list_ports.comports
12:45:54.655 INFO    thonny.plugins.micropython.mp_front: Falling back to serial.tools.list_ports.comports
12:45:54.903 INFO    thonny.ui_utils: Parent y: 58, rooty: 109, vrooty: 0
12:45:54.904 INFO    thonny.ui_utils: Placing .!configurationdialog with geometry +860+213
12:46:03.953 INFO    thonny.ui_utils: Parent y: 58, rooty: 109, vrooty: 0
12:46:03.953 INFO    thonny.ui_utils: Placing .!nodechoicedialog with geometry +1025+376
12:46:05.252 INFO    thonny.ui_utils: Parent y: 58, rooty: 109, vrooty: 0
12:46:05.253 INFO    thonny.ui_utils: Placing .!backendfiledialog with geometry +868+178
12:46:13.406 INFO    thonny.ui_utils: Parent y: 58, rooty: 109, vrooty: 0
12:46:13.406 INFO    thonny.ui_utils: Placing .!inlinecommanddialog with geometry +984+406
12:46:20.634 INFO    thonny.ui_utils: Parent y: 58, rooty: 109, vrooty: 0
12:46:20.635 INFO    thonny.ui_utils: Placing .!inlinecommanddialog2 with geometry +984+406
12:46:21.021 INFO    thonny.running: Reader got EOF
12:46:21.162 INFO    thonny.shell: BaseShellText._on_backend_terminated

```

I'm currently wading through the python code to locate where "write_file" is called from.

Odd that Thonny opens a file from the com port associated with instance but doesn't know where to put it when saving. Note this only happens with more than once instance of Thonny open. I'll have to use something else to work around this problem.

BNNorman, 2022-10-23:
A friend reckoned his v3.3.3 worked so I downloaded that and installed it.
It allows me to run code on two instances BUT if I try to save then I get the same error message. IIRC that was the reason I tried a later version. However 3.3.13 & 4.0.1 seems to want to save the file before running.

I am using default settings except selecting "Allow only single Thonny instance".

I can edit the code and save outside of Thonny as a work around.

aivarannamaa, 2022-10-29:
Thank you for the report!

I assume you tried to save your files as "remote" files, not "local" files. In this case Thonny knows the port the device is connected to but there is no explicit link available for learning the correct mount point of the device's filesystem (this is required only with CircuitPython). Normally Thonny can infer it, but with two devices attached, it can't choose.

There are some enhanced planned for such situations, but in your case you could try saving directly on the correct drive via Thonny's "local" save option.

BNNorman, 2022-10-29:
Thanks. As I said, I can work around it. It was save to remote that fails. I would have thought that storing the full file path, when opened, would solve the problem with circuitpython. If I get a moment I may look over the sources.

closed 2023-01-28 by aivarannamaa