thonny/thonny #2516: Permission denied when starting Thonny 4.0.1 - THONNY_USER_DIR

feiticeir0, 2022-10-22:
Hi ! 
I'm having an issue with Thonny. 

When trying to start it, I get the following error:
`[feiticeir0@Nightcrawler ~]$ thonny 
Traceback (most recent call last):
  File "/home/feiticeir0/.local/bin/thonny", line 8, in <module>
    sys.exit(launch())
  File "/home/feiticeir0/.local/lib/python3.10/site-packages/thonny/__init__.py", line 227, in launch
    prepare_thonny_user_dir()
  File "/home/feiticeir0/.local/lib/python3.10/site-packages/thonny/__init__.py", line 273, in prepare_thonny_user_dir
    os.makedirs(THONNY_USER_DIR, mode=0o700, exist_ok=True)
  File "/usr/lib/python3.10/os.py", line 225, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/usr/.thonny'
`
I'm using Arch Linux. 
This version was installed using PIP, but with others installed via AUR it's the same problem. 

This only happens with version 4. Versions 3 were working correctly. 
I'm using Python 3.10.8

What could it be ? Thank you

aivarannamaa, 2022-10-29:
Thank you for the report!

What does following print on your computer:

```python
>>> import os.path
>>> os.path.expanduser("~/")
```

feiticeir0, 2022-10-30:
Thanks for the reply

> Thank you for the report!
> 
> What does following print on your computer:
> 
> ```python
> >>> import os.path
> >>> os.path.expanduser("~/")
> ```

it returns
`>>> import os.path
>>> os.path.expanduser("~/")
'/home/feiticeir0/'
`

aivarannamaa, 2022-10-30:
For some reason it looks like Thonny thinks `/home/feiticeir0/.local/bin/` is a virtual environment... Is it? Do you have a file named `activate` in this folder?

feiticeir0, 2022-10-30:
I have not. 
I've build Cura from source and have conan installed ... could it be it ?  I have some Python packages here - binary

aivarannamaa, 2022-12-28:
Unfortunately I don't know how to debug this situation further. Please re-open this issue when you have more information to add.

closed 2022-12-28 by aivarannamaa