thonny/thonny #2517: Install from local file section doesn't work ( Tools -> Manage plug-ins -> Install from local file section )

HighASG936, 2022-10-23:
Traceback (most recent call last):
  File "/usr/lib/python3.8/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/usr/lib/python3/dist-packages/thonny/plugins/pip_gui.py", line 655, in _handle_install_file_click
    filename = askopenfilename(
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1920, in askopenfilename
    return _get_dialog_provider().askopenfilename(**options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1960, in askopenfilename
    args = cls._convert_common_options("Open file", **options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 1994, in _convert_common_options
    filename = _options_to_zenity_filename(options)
  File "/usr/lib/python3/dist-packages/thonny/ui_utils.py", line 2036, in _options_to_zenity_filename
    return options["initialdir"] + os.path.sep
TypeError: unsupported operand type(s) for +: 'method' and 'str'

aivarannamaa, 2022-10-29:
Thank you for the report!

Which Thonny version are you using?

HighASG936, 2022-11-05:
Thonny 4.0.1

aivarannamaa, 2022-11-06:
It doesn't look right. The line numbers in the stacktrace doesn't match with the ones in Thonny 4.0.1. Also, I can't reproduce this problem in 4.0.1 but I remember having such a bug in Thonny 3.3.

Please re-check your Thonny version.

aivarannamaa, 2022-12-30:
Please re-open if you are sure you are using Thonny 4.0.1 (e.g. by checking from "Help => About")

closed 2022-12-30 by aivarannamaa