thonny/thonny #2520: PROBLEM IN THONNY'S BACK-END

Darklight29, 2022-10-24:
PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.

aivarannamaa, 2022-10-29:
Thank you for the report!

I haven't met this error before. Did it happen once or can you reproduce it consistently? How?

Which OS are you using?

Darklight29, 2022-10-30:
I get it constantly.

I downloaded https://www.raspberrypi.com/documentation/microcontrollers/micropython.html from Raspberry Pi.org. Installed it to the Pico by dragging and dropping onto the Pico.

I'm running Thonny on a windows 10 pc.
The Pico is running MicroPython (Raspberry Pi Pico)
If I go to local Python 3 I can run the print hi program print ;print ('Hi world')

If I choose MicroPython(Raspberry Pi Pico), it gives me the error and goes into a loop.

aivarannamaa, 2022-10-30:
Please see the instructions for connecting to your Pico with Putty. Can you do this successfully?

Darklight29, 2022-10-31:
I don't have those instructions. I did plug it in and neither of my scanners pick it up.

aivarannamaa, 2022-10-31:
Are you sure you are connecting to the right port? Maybe this can help you find out: https://picockpit.com/raspberry-pi/raspberry-pi-pico-and-micropython-on-windows/

jberes-ae, 2022-11-02:
i echo darklight29's issue. we, too, are continuously experiencing the error below. we can no longer develop our device. we are running windows10-home version and 21h1 and windows10-pro 21h2 version machines.  we are developing an rp2040 device with MicroPython v1.19.1 on 2022-06-18.

we would appreciate guidance on how to work-around this issue. 

thank you

********
PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_dirs_children_info' (serial.serialutil.SerialTimeoutException: Write timeout).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.

aivarannamaa, 2022-11-06:
@jberes-ae, can you connect to your Pico with Putty?

breaghva, 2022-11-19:
I may have the same issue.  Pico W, simple micropython program to control WS1812 list strip.  latest version of Thonny

22:34:03.326 INFO    thonny: sys.executable: /usr/bin/python3
22:34:03.326 INFO    thonny: sys.argv: ['/usr/lib/python3/dist-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy>
22:34:03.327 INFO    thonny: sys.path: ['/usr/lib/python3/dist-packages/thonny/plugins/rp2040', '/usr/lib/python39.zip', '/usr/lib/python3.9', '/usr/lib/python3.9/lib-dynload', '/home/pi/.local/lib/python3.9/site-packages', '/usr/local/lib/python3.9/dist-packages', '/usr/lib/python3/dist-pack>
22:34:03.327 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0)
22:34:03.329 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryP>
22:34:08.372 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 64, write_block_delay: 0, read_block_size: 0
22:34:08.373 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type RP2040Backend
22:34:08.378 INFO    thonny: TIME/MODS 5.334 before prepare (+76 modules)
22:34:08.378 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'ctypes', 'ctypes._endian', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch'>
22:34:08.393 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:34:10.396 ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/mp_back.py", line 139, in __init__
    self._process_until_initial_prompt(
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 268, in _process_until_initial_prompt
    self._process_output_until_active_prompt(
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 844, in _process_output_until_active_prompt
    self._interrupt()
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/mp_back.py", line 337, in _interrupt
    self._write(INTERRUPT_CMD)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/mp_back.py", line 466, in _write
    return self._connection.write(data)
  File "/usr/lib/python3/dist-packages/thonny/plugins/micropython/serial_connection.py", line 89, in write
    size = self._serial.write(data)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 618, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

aivarannamaa, 2022-12-28:
@breaghva, your problem seems different. I assume you are using Unix variant of MicroPython. If your problem persists then please create a separate issue.

For the others -- there is no point for me to debug it further before you can tell me whether you can use your board properly with another tool (e.g. Putty)

closed 2022-12-28 by aivarannamaa

RufusVS, 2023-01-21:
I am experiencing this problem with Thonny.  TeraTerm connect fine.

ozoidemi, 2023-08-02:
I just hit the original  error working with a raspberry pi pico w.

> PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))).
> See Thonny's [...]

Running Windows 11 Education 22H2 and Thonny 4.1.1.

Putty cannot connect either. 

> [Retry #286] Waiting for connection...
> [Retry #287] Waiting for connection...
> [Retry #288] Waiting for connection...
> [Retry #289] Waiting for connection...
> [Retry #290] Waiting for connection...
> [Retry #291] Waiting for connection...

Problem happened after running ([this code](https://projects.raspberrypi.org/en/projects/get-started-pico-w/2)).

`import network
import socket
from time import sleep
from picozero import pico_temp_sensor, pico_led
import machine

ssid = 'NAME OF YOUR WIFI NETWORK'
password = 'YOUR SECRET PASSWORD'

def connect():
    #Connect to WLAN
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(ssid, password)
    while wlan.isconnected() == False:
        print('Waiting for connection...')
        sleep(1)
    print(wlan.ifconfig())

try:
    connect()
except KeyboardInterrupt:
    machine.reset()
`
Specifically, the issue began with the `KeyboadInterrupt` exception. I CTRL+C my pi pico, after it was unable to connect, and then completely lost it.

spragu, 2023-10-07:
"PROBLEM IN THONNY'S BACK-END"
I dont think there will ever be a point in my life where i'll be mature enough to not lol at that.

JuviGamez, 2024-02-03:
I Have Faced The Same Issue While Using A Raspberry Pi Pico, Like How ozoidemi Said If You Hit CTRL+C The Error Starts

ozoidemi, 2024-04-19:
> Stumbled upon the same problem, commenting out except KeyboardInterupt solved it, as @ozoidemi said (thanks for the hint!) but is that a workaround or the real solution ?

Glad that the solution worked for you.All I can say is that I didn’t have any other issues after nuking the pico, though the final version of the code wasn’t much more complicated than what was posted.Without an in-depth analysis, I’d find it challenging to say whether this is a workaround or not.Also, since no further problems seemed to appear after nuking, not for me or anyone else using this solution (I mean, they might have said something if they had seen other related issues, right?), then it is a question of whether spending time to figure this problem out is really the best use of that time.

Squeek73, 2024-08-13:
Same stuff: Problem in Thonny's back end.

- Windows 11, Thonny 4.1.4
MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040

The Pico has main.py on it, which loads automatically on powerup
Thonny is open, and it can't connect via USB
Nuke the thing
Drop RPI_PICO_W on it
Click on stop/restart in Thonny and I get "MPY: soft reboot"
Click open from the pico, no files, of course, but I have a backup copy of main.py stored as code.py on my computer, so I load that, and it runs fine.
MicroPython (RP2040) Board CDC @ com8

The code connects to wifi and serves a page with a 3x3 matrix of buttons for
test  forward  test
left  stop  right
test  backward  test

so I can move my "tank" via wifi.  It's all working fine now.

Click stop/restart and it goes back into that 
"PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22)))."

Close and reopen Thonny, and it's back to normal, running the code
Click stop/restart and it gives that error again.  Click it again and it does the soft reboot.  Click RUN, and the code runs fine.

Save the code as main.py on the Pico and stop/run and you can't stop it with stop/restart.  

Save the code to main.py on the Pico and power cycle.
It auto-runs as expected, it gets an IP adress, the web page loads, it responds to clicks fine.  Click stop/restart and it can't be stopped in Thonny, but the code sure stopped.

Power cycle, same thing - it works and stop/restart kills it.

I can reproduce it - nuke/micropython/thonny/load code/run, stop/restart - BOOM

I think it might have something to do with it running main.py as if it's actually still running somethin gor other, and Thonny isn't fast enough to break in.

That's my story and I'm sticking to it.