thonny/thonny #2522: Error code 17 after every

firefoxlover, 2022-10-26:
I get this after every executed command (everything works normally:)

```
PROBLEM IN THONNY'S BACK-END: mainloop error (FileNotFoundError: [Errno 2] No such file or directory).
See Thonny's backend.log for more info.

Process ended with exit code 17.
```

I use Thonny with Shell, Assistant, Scriptwindow and Variables displayed. This is the backend.log:

```
20:39:25.159 INFO    thonny: Thonny version: 4.0.1
20:39:25.159 INFO    thonny: cwd: /app/lib/python3.10/site-packages
20:39:25.159 INFO    thonny: original argv: ['/usr/bin/python3', '-u', '-B', '-m', 'thonny.plugins.cpython_backend.cp_launcher', '/run/flatpak/doc/778c89df']
20:39:25.160 INFO    thonny: sys.executable: /usr/bin/python3
20:39:25.160 INFO    thonny: sys.argv: ['/app/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_launcher.py', '/run/flatpak/doc/778c89df']
20:39:25.160 INFO    thonny: sys.path: ['/app/lib/python3.10/site-packages', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/app/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages']
20:39:25.160 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
20:39:25.161 INFO    thonny: TIME/MODS 0.003 Before constructing backend (+2 modules)
20:39:25.161 INFO    thonny: NEW MODS ['encodings.ascii', 'faulthandler']
20:39:25.161 INFO    thonny: TIME/MODS 0.000 Before MainBackend 
20:39:25.161 INFO    thonny: TIME/MODS 0.000 After MainBackend 
20:39:25.177 INFO    thonny.backend: Loading plugins
20:39:25.177 INFO    thonny: TIME/MODS 0.016 Before loading plugins (+3 modules)
20:39:25.177 INFO    thonny: NEW MODS ['pkgutil', 'platform', 'pydoc']
20:39:25.177 INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/app/bin', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/app/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages', '/home/user/.var/app/org.thonny.Thonny/config/Thonny/plugins/lib/python3.10/site-packages']
20:39:25.179 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.birdseye_backend'
20:39:25.180 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.dock_user_windows_backend'
20:39:25.181 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.flask_backend'
20:39:25.182 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.matplotlib_backend'
20:39:25.183 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.pgzero_backend'
20:39:25.184 INFO    thonny: TIME/MODS 0.007 After loading plugins (+6 modules)
20:39:25.184 INFO    thonny: NEW MODS ['thonny.plugins.backend', 'thonny.plugins.backend.birdseye_backend', 'thonny.plugins.backend.dock_user_windows_backend', 'thonny.plugins.backend.flask_backend', 'thonny.plugins.backend.matplotlib_backend', 'thonny.plugins.backend.pgzero_backend']
20:39:25.184 INFO    thonny: TIME/MODS 0.001 Beginning of mainloop 
20:39:25.185 INFO    thonny: TIME/MODS 0.001 Sending message ToplevelResponse 
20:39:25.186 INFO    thonny.backend: Loading Jedi
20:39:25.186 INFO    thonny: TIME/MODS 0.001 Before loading Jedi 
20:39:25.187 INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/app/bin', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/app/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages', '/home/user/.var/app/org.thonny.Thonny/config/Thonny/plugins/lib/python3.10/site-packages']
20:39:25.414 INFO    thonny: TIME/MODS 0.228 After loading Jedi (+126 modules)
20:39:25.415 INFO    thonny: NEW MODS ['__future__', '_blake2', '_bz2', '_compat_pickle', '_compression', '_hashlib', '_json', '_lzma', '_pickle', '_struct', 'binascii', 'bz2', 'difflib', 'filecmp', 'gc', 'hashlib', 'jedi', 'jedi._compatibility', 'jedi.api', 'jedi.api.classes', 'jedi.api.completion', 'jedi.api.completion_cache', 'jedi.api.environment', 'jedi.api.errors', 'jedi.api.exceptions', 'jedi.api.file_name', 'jedi.api.helpers', 'jedi.api.interpreter', 'jedi.api.keywords', 'jedi.api.project', 'jedi.api.refactoring', 'jedi.api.refactoring.extract', 'jedi.api.strings', 'jedi.cache', 'jedi.common', 'jedi.debug', 'jedi.file_io', 'jedi.inference', 'jedi.inference.analysis', 'jedi.inference.arguments', 'jedi.inference.base_value', 'jedi.inference.cache', 'jedi.inference.compiled', 'jedi.inference.compiled.access', 'jedi.inference.compiled.getattr_static', 'jedi.inference.compiled.mixed', 'jedi.inference.compiled.subprocess', 'jedi.inference.compiled.subprocess.functions', 'jedi.inference.compiled.value', 'jedi.inference.context', 'jedi.inference.docstring_utils', 'jedi.inference.docstrings', 'jedi.inference.filters', 'jedi.inference.flow_analysis', 'jedi.inference.gradual', 'jedi.inference.gradual.annotation', 'jedi.inference.gradual.base', 'jedi.inference.gradual.conversion', 'jedi.inference.gradual.generics', 'jedi.inference.gradual.stub_value', 'jedi.inference.gradual.type_var', 'jedi.inference.gradual.typeshed', 'jedi.inference.gradual.typing', 'jedi.inference.gradual.utils', 'jedi.inference.helpers', 'jedi.inference.imports', 'jedi.inference.lazy_value', 'jedi.inference.names', 'jedi.inference.param', 'jedi.inference.parser_cache', 'jedi.inference.recursion', 'jedi.inference.references', 'jedi.inference.signature', 'jedi.inference.syntax_tree', 'jedi.inference.sys_path', 'jedi.inference.utils', 'jedi.inference.value', 'jedi.inference.value.decorator', 'jedi.inference.value.dynamic_arrays', 'jedi.inference.value.function', 'jedi.inference.value.instance', 'jedi.inference.value.iterable', 'jedi.inference.value.klass', 'jedi.inference.value.module', 'jedi.parser_utils', 'jedi.plugins', 'jedi.plugins.django', 'jedi.plugins.flask', 'jedi.plugins.pytest', 'jedi.plugins.registry', 'jedi.plugins.stdlib', 'jedi.settings', 'json', 'json.decoder', 'json.encoder', 'json.scanner', 'lzma', 'parso', 'parso._compatibility', 'parso.cache', 'parso.file_io', 'parso.grammar', 'parso.normalizer', 'parso.parser', 'parso.pgen2', 'parso.pgen2.generator', 'parso.pgen2.grammar_parser', 'parso.python', 'parso.python.diff', 'parso.python.errors', 'parso.python.parser', 'parso.python.pep8', 'parso.python.prefix', 'parso.python.token', 'parso.python.tokenize', 'parso.python.tree', 'parso.tree', 'parso.utils', 'pickle', 'pydoc_data', 'pydoc_data.topics', 'shutil', 'struct', 'textwrap', 'zipfile', 'zlib']
```

Operating System: Fedora Linux 36
KDE-Plasma-Version: 5.25.5
KDE-Frameworks-Version: 5.99.0
Qt-Version: 5.15.6
Kernel-Version: 5.19.16-200.fc36.x86_64 (64-bit)
Grafics-Platform: X11

aivarannamaa, 2022-10-29:
Thank you for the report!

Unfortunately the log doesn't mention the error. Please note that the backend.log gets cleared each time Thonny starts a new backend process. It's best to close Thonny right after experiencing the error and then check the log file.

firefoxlover, 2022-11-01:
```
10:43:01.509 INFO    thonny: Thonny version: 4.0.1
10:43:01.510 INFO    thonny: cwd: /app/lib/python3.10/site-packages
10:43:01.510 INFO    thonny: original argv: ['/usr/bin/python3', '-u', '-B', '-m', 'thonny.plugins.cpython_backend.cp_launcher', '/run/flatpak/doc/f577d41c']
10:43:01.510 INFO    thonny: sys.executable: /usr/bin/python3
10:43:01.510 INFO    thonny: sys.argv: ['/app/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_launcher.py', '/run/flatpak/doc/f577d41c']
10:43:01.510 INFO    thonny: sys.path: ['/app/lib/python3.10/site-packages', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/app/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages']
10:43:01.510 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
10:43:01.510 INFO    thonny: TIME/MODS 0.002 Before constructing backend (+2 modules)
10:43:01.511 INFO    thonny: NEW MODS ['encodings.ascii', 'faulthandler']
10:43:01.511 INFO    thonny: TIME/MODS 0.000 Before MainBackend 
10:43:01.511 INFO    thonny: TIME/MODS 0.000 After MainBackend 
10:43:01.519 INFO    thonny.backend: Loading plugins
10:43:01.519 INFO    thonny: TIME/MODS 0.008 Before loading plugins (+3 modules)
10:43:01.519 INFO    thonny: NEW MODS ['pkgutil', 'platform', 'pydoc']
10:43:01.519 INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/app/bin', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/app/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages', '/home/user/.var/app/org.thonny.Thonny/config/Thonny/plugins/lib/python3.10/site-packages']
10:43:01.520 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.birdseye_backend'
10:43:01.521 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.dock_user_windows_backend'
10:43:01.521 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.flask_backend'
10:43:01.522 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.matplotlib_backend'
10:43:01.522 INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.pgzero_backend'
10:43:01.523 INFO    thonny: TIME/MODS 0.004 After loading plugins (+6 modules)
10:43:01.523 INFO    thonny: NEW MODS ['thonny.plugins.backend', 'thonny.plugins.backend.birdseye_backend', 'thonny.plugins.backend.dock_user_windows_backend', 'thonny.plugins.backend.flask_backend', 'thonny.plugins.backend.matplotlib_backend', 'thonny.plugins.backend.pgzero_backend']
10:43:01.523 INFO    thonny: TIME/MODS 0.000 Beginning of mainloop 
10:43:01.523 INFO    thonny: TIME/MODS 0.000 Sending message ToplevelResponse 
10:43:01.524 INFO    thonny.backend: Loading Jedi
10:43:01.524 INFO    thonny: TIME/MODS 0.000 Before loading Jedi 
10:43:01.524 INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/app/bin', '/usr/lib/python310.zip', '/usr/lib/python3.10', '/usr/lib/python3.10/lib-dynload', '/app/lib/python3.10/site-packages', '/usr/lib/python3.10/site-packages', '/home/user/.var/app/org.thonny.Thonny/config/Thonny/plugins/lib/python3.10/site-packages']
10:43:01.620 INFO    thonny: TIME/MODS 0.096 After loading Jedi (+126 modules)
10:43:01.620 INFO    thonny: NEW MODS ['__future__', '_blake2', '_bz2', '_compat_pickle', '_compression', '_hashlib', '_json', '_lzma', '_pickle', '_struct', 'binascii', 'bz2', 'difflib', 'filecmp', 'gc', 'hashlib', 'jedi', 'jedi._compatibility', 'jedi.api', 'jedi.api.classes', 'jedi.api.completion', 'jedi.api.completion_cache', 'jedi.api.environment', 'jedi.api.errors', 'jedi.api.exceptions', 'jedi.api.file_name', 'jedi.api.helpers', 'jedi.api.interpreter', 'jedi.api.keywords', 'jedi.api.project', 'jedi.api.refactoring', 'jedi.api.refactoring.extract', 'jedi.api.strings', 'jedi.cache', 'jedi.common', 'jedi.debug', 'jedi.file_io', 'jedi.inference', 'jedi.inference.analysis', 'jedi.inference.arguments', 'jedi.inference.base_value', 'jedi.inference.cache', 'jedi.inference.compiled', 'jedi.inference.compiled.access', 'jedi.inference.compiled.getattr_static', 'jedi.inference.compiled.mixed', 'jedi.inference.compiled.subprocess', 'jedi.inference.compiled.subprocess.functions', 'jedi.inference.compiled.value', 'jedi.inference.context', 'jedi.inference.docstring_utils', 'jedi.inference.docstrings', 'jedi.inference.filters', 'jedi.inference.flow_analysis', 'jedi.inference.gradual', 'jedi.inference.gradual.annotation', 'jedi.inference.gradual.base', 'jedi.inference.gradual.conversion', 'jedi.inference.gradual.generics', 'jedi.inference.gradual.stub_value', 'jedi.inference.gradual.type_var', 'jedi.inference.gradual.typeshed', 'jedi.inference.gradual.typing', 'jedi.inference.gradual.utils', 'jedi.inference.helpers', 'jedi.inference.imports', 'jedi.inference.lazy_value', 'jedi.inference.names', 'jedi.inference.param', 'jedi.inference.parser_cache', 'jedi.inference.recursion', 'jedi.inference.references', 'jedi.inference.signature', 'jedi.inference.syntax_tree', 'jedi.inference.sys_path', 'jedi.inference.utils', 'jedi.inference.value', 'jedi.inference.value.decorator', 'jedi.inference.value.dynamic_arrays', 'jedi.inference.value.function', 'jedi.inference.value.instance', 'jedi.inference.value.iterable', 'jedi.inference.value.klass', 'jedi.inference.value.module', 'jedi.parser_utils', 'jedi.plugins', 'jedi.plugins [... cut ...] rso._compatibility', 'parso.cache', 'parso.file_io', 'parso.grammar', 'parso.normalizer', 'parso.parser', 'parso.pgen2', 'parso.pgen2.generator', 'parso.pgen2.grammar_parser', 'parso.python', 'parso.python.diff', 'parso.python.errors', 'parso.python.parser', 'parso.python.pep8', 'parso.python.prefix', 'parso.python.token', 'parso.python.tokenize', 'parso.python.tree', 'parso.tree', 'parso.utils', 'pickle', 'pydoc_data', 'pydoc_data.topics', 'shutil', 'struct', 'textwrap', 'zipfile', 'zlib']

```

Viizzo67, 2022-12-04:
> edited it

edit what file. can you post the path and what to do?

firefoxlover, 2022-12-04:
@Viizzo67 This error was always occuring using the Flatpak version on Fedora 37. I edited the original issue report and integrated the log, if there is no issue visible I dont know. I cant reproduce it at the moment using Fedora Kionite 37

aivarannamaa, 2022-12-05:
Unfortunately I have no idea how to start solving this. I'm going to close the issue for now, but feel free to reopen it if you get more leads.

closed 2022-12-05 by aivarannamaa