thonny/thonny #2523: Thonny 3.3.10 Not Loading or Saving

trex101, 2022-10-26:
While coding in Thonny Python IDE, I tried to control RPi’s alsa audio output using a shell command “os.system()” which apparently caused Thonny to lose its ability to Load or Save programs. I now have to use Geany as a fallback, but would like to restore these functions back in Thonny - any suggestions on how to do this would be greatly appreciated. I backed up the SD card’s image but it has the same Thonny issues; I’ve tried updating/reloading Thonny (sudo apt-get remove/purge python/3-thonny), these remove not just Thonny but dozens if not hundreds of other programs which is unacceptable. Wondering if there’s a way to remove/reinstall specific drivers for Thonny that handle file transfers. I’m running Raspian Linux 10 / Raspberry Pi OS Buster and Thonny 3.3.10 with Python 3.7.3 (32 bit) Tk 8.6.9.

aivarannamaa, 2022-10-29:
Thank you for the report!

> I tried to control RPi’s alsa audio output using a shell command “os.system()” which apparently caused Thonny to lose its ability to Load or Save programs.

What do you mean by losing the ability to load or save programs? Do you get an error message? Or nothing happens when you try to save/load? If nothing happens, then you could try "Tools => Options => General => Use Tk file dialogs instead of Zenity".

trex101, 2023-01-09:
> Thank you for the report!
>
>
> What do you mean by losing the ability to load or save programs? Do you get an error message? Or nothing happens when you try to save/load? If nothing happens, then you [...]

Thank you! That worked!

closed 2023-01-09 by trex101