thonny/thonny #2524: micropython-umqtt.simple does not install

artspace133, 2022-10-29:
The micropython-umqtt.simple packace does not install correctly.
The version is 1.3.4

[image]

aivarannamaa, 2022-10-29:
Thank you for the report!

Is this also your message: https://groups.google.com/g/thonny/c/0jtdT07nPYQ ?

artspace133, 2022-10-29:
Thank You,
The message from the google groups is from another user.

mbradley612, 2022-11-26:
Follow the instruction here to install these libraries on a RPi Pico using Thonny: https://www.hivemq.com/blog/iot-reading-sensor-data-raspberry-pi-pico-w-micropython-mqtt-node-red/. You'll need to install both umqtt.simple and umqtt.robust.

'''
import upip
upip.install(“micropython-umqtt.robust”)
upip.install(“micropython-umqtt.simple”)
'''

artspace133, 2022-11-28:
I have had no luck trying to install the work around mentioned in this blog
post. I'm stuck until this is fixed...
Thanks to all, please let me know if anyone has got this to work...

jesper-johansen, 2022-11-28:
It looks as upip is replaced with mip, so for me the following is working
>>>import mip
>>>mip.install('umqtt.simple')
>>>mip.install('umqtt.robust')
regards Jesper

artspace133, 2022-11-28:
When I tried to install "mip", a message indicated that the package is
intended for circuit python use, and it failed...
I'm using a Pico W... Sorry to all that I didn't mention that...

Justin Parker

GAZDOWN, 2022-11-29:
> It looks as upip is replaced with mip, so for me the following is working
> 
> > > > import mip
> > > > mip.install('umqtt.simple')
> > > > mip.install('umqtt.robust')
> > > > regards Jesper

Hi I can confirm this indeed works.

marcelomaiajuvencio, 2022-12-05:
Hello!!!
How to solve:
download the micropython-umqtt.simple-1.3.4.tar package at [https://pypi.org/project/micropython-umqtt.simple/#files](https://pypi.org/project/micropython-umqtt.simple/#files)
[image]

Extract the micropython-umqtt.simple-1.3.4.tar package
[image]

Using Thonny, with Pico connected, navigate to the umqtt folder
[image]

Right click on the umqtt folder > Upload to/
[image]

The folder will be copied on Pico
[image]

Ready, just import and use
`from umqtt.simple import MQTTClient`

artspace133, 2022-12-05:
Thank you very much, I finally got this working on my Pick W...

Thanks again,
Justin Parker

aivarannamaa, 2022-12-28:
I fixed the root cause in pipkin (https://github.com/aivarannamaa/pipkin/issues/4) and the fix will be published as part of Thonny 4.0.4

aivarannamaa, 2022-12-31:
Fixed by 79625901

closed 2022-12-31 by aivarannamaa

marcelomaiajuvencio, 2023-01-11:
Hello, I tried with mip but it didn't work either. So I adopted the manual installation procedure.

 
 

Hi-
tried the mip approach

Installing umqtt.simple (latest) from https://micropython.org/pi/v2 to /lib
Traceback (most recent call last):
File "", line 1, in 
File "mip/init.py", line 1, in install
File "mip/init.py", line 1, in _install_package
File "mip/init.py", line 1, in _install_json
File "urequests.py", line 180, in get
File "urequests.py", line 76, in request
OSError: -6

Is there something I should check on my side?
Thx.

tpilutti, 2023-01-27:
Hi - anyone know if Issue 2524 was resolved in Thonny release 4.0.2? Didn't see this issue number in the 4.0.2 release notes.  Thanks.

aivarannamaa, 2023-01-27:
Yes, this should be fixed in 4.0.2.

tpilutti, 2023-01-29:
Thanks! ManagePackages tool in 4.0.2 looks like it installed successfully. 

[image]

ElToro1966, 2023-03-01:
On Thonny 3.2.7:
>>> import mip
>>> mip.install("umqtt.simple")
Installing umqtt.simple (latest) from https://micropython.org/pi/v2 to /lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mip/__init__.py", line 1, in install
  File "mip/__init__.py", line 1, in _install_package
  File "mip/__init__.py", line 1, in _install_json
  File "urequests.py", line 180, in get
  File "urequests.py", line 76, in request
OSError: -6

However:
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.connect('wifi_ssid', 'wifi_pw')
>>> import mip
>>> mip.install("umqtt.simple")
Installing umqtt.simple (latest) from https://micropython.org/pi/v2 to /lib
Copying: /lib/umqtt/simple.mpy
Done
>>>

ArPiSo, 2023-07-12:
[image]

C:\Users\ПК\AppData\Local\Programs\Thonny\python.exe -m pip --disable-pip-version-check --no-color install --no-warn-script-location --user --user --upgrade micropython-umqtt.simple
Requirement already satisfied: micropython-umqtt.simple in c:\users\РїРє\appdata\roaming\thonny\plugins\python310\site-packages (1.1)
Collecting micropython-umqtt.simple
  Using cached micropython-umqtt.simple-1.3.4.tar.gz (4.0 kB)
ERROR: micropython-umqtt.simple from https://files.pythonhosted.org/packages/bd/cf/697e3418b2f44222b3e848078b1e33ee76aedca9b6c2430ca1b1aec1ce1d/micropython-umqtt.simple-1.3.4.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Error: process returned with code 1

aivarannamaa, 2023-07-23:
@ArPiSo, it looks like you were trying to install a MicroPython package as Thonny plug-in. Please read the text on the top of your screenshot.