thonny/thonny #2526: problem after install when open the program

jamalshehri, 2022-10-30:
[image]
what is the problem  and how to solve it ?

aivarannamaa, 2022-10-30:
I see that you are using Windows 7. Which Thonny version are you using? The one with Python 3.10 or 3.8 ?

jamalshehri, 2022-11-01:
its 4.0.1

aivarannamaa, 2022-11-01:
Right, but did you download https://github.com/thonny/thonny/releases/download/v4.0.1/thonny-4.0.1.exe or https://github.com/thonny/thonny/releases/download/v4.0.1/thonny-py38-4.0.1.exe ? The former does not work on Windows 7.

jamalshehri, 2022-11-01:
I try the both and its the same result after installation

aivarannamaa, 2022-11-01:
Unfortunately I can't reproduce this, but let's try the fix that works for [someone with similar problem](https://github.com/marcelotduarte/cx_Freeze/issues/523).

Please try following:

* Locate Thonny's installation directory. According to your screenshot it is `C:\Users\PC.SCHOOL\AppData\Local\Programs\Thonny`. The directory `C:\Users\PC.SCHOOL\AppData` may be hidden, so you need to either make all files visible, or in Windows Explorer paste `C:\Users\PC.SCHOOL\AppData\Local\Programs\Thonny` to the "address bar" and press ENTER.
* Make sure you see the directories `DLLs` and `Lib` under Thonny directory
* From `DLLs` copy the file named `libffi-7.dll` to `Lib` directory.

If this doesn't help and you still get the same error message, the please copy `libffi-7.dll` to Thonny's main directory instead, and try again.

jamalshehri, 2022-11-01:
thanks for ur help I will try and feedback again :)

closed 2023-01-28 by aivarannamaa