thonny/thonny #2527: Problem when opening Thonny

yubub, 2022-10-30:
hi I'm using macOS Monterey 12.3.1 and Thonny 4.0.1.
I'm getting this message every time i open Thonny and try to work with micropython, with python it dosn't show.
thanx

PROBLEM IN THONNY'S BACK-END: Internal error (thonny.plugins.micropython.mp_back.ManagementError: Script produced errors).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.

aivarannamaa, 2022-10-30:
Thank you for the report!

After you get this error again, please select "Tools => Open Thonny data folder" and send me the backend.log from this folder.

yubub, 2022-11-13:
17:22:42.659 INFO    thonny: Thonny version: 4.0.1
17:22:42.660 INFO    thonny: cwd: /
17:22:42.660 INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': False, 'port': '/dev/cu.usbmodem14601', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
17:22:42.660 INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
17:22:42.660 INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': False, 'port': '/dev/cu.usbmodem14601', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
17:22:42.660 INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Users/yulitamir/Library/Python/3.10/lib/python/site-packages', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
17:22:42.660 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
17:22:42.660 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': '/dev/cu.usbmodem14601', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
17:22:42.664 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 64, write_block_delay: 0, read_block_size: 0
17:22:42.665 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type RP2040Backend
17:22:42.665 INFO    thonny: TIME/MODS 0.080 before prepare (+73 modules)
17:22:42.665 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
17:22:42.666 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
17:22:42.666 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
17:22:42.691 INFO    thonny: TIME/MODS 0.026 got welcome 
17:22:42.691 INFO    thonny: TIME/MODS 0.000 bef preparing helpers 
17:22:42.691 INFO    thonny.plugins.micropython.mp_back: Preparing helpers
17:22:42.692 INFO    thonny: TIME/MODS 0.000 befsubcode 
17:22:42.692 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
17:22:42.703 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
17:22:42.742 INFO    thonny: TIME/MODS 0.051 affsubcode 
17:22:42.833 INFO    thonny: TIME/MODS 0.091 affforw 
17:22:42.834 INFO    thonny: TIME/MODS 0.000 prepared helpers 
17:22:42.834 INFO    thonny: TIME/MODS 0.000 befsubcode 
17:22:42.836 INFO    thonny: TIME/MODS 0.002 affsubcode 
17:22:42.848 INFO    thonny: TIME/MODS 0.012 affforw 
17:22:42.848 INFO    thonny: TIME/MODS 0.000 got cwd 
17:22:42.848 INFO    thonny: TIME/MODS 0.000 befsubcode 
17:22:42.851 INFO    thonny: TIME/MODS 0.002 affsubcode 
17:22:42.865 INFO    thonny: TIME/MODS 0.014 affforw 
[... cut ...]
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 571, in _evaluate
    raise ManagementError("Script produced errors", script, out, err)
thonny.plugins.micropython.mp_back.ManagementError: Script produced errors
17:22:42.866 ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: __thonny_helper.print_mgmt_value(__thonny_helper.sys.path)

STDOUT: 

STDERR: Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'path'

aivarannamaa, 2022-11-14:
Which device are you using? Which version of MicroPython?

yubub, 2022-11-15:
Regular raspberry pi pico and the version of micropython for pico.

aivarannamaa, 2022-11-16:
The error log seems to say that it can't access field `path` in the `sys` module on your device. This is strange, because almost all MicroPython variants I've seen (except micro:bit's) have `sys.path`.

Can you connect to MicroPython REPL with a simpler tool (eg. `screen`) and check whether you can execute following code:

```python
import sys
print(sys.path)
```
?

Do you happen to have a file named `sys.py` on your device?

yubub, 2022-11-17:
Yes, I've found it.

yubub, 2022-11-18:
ERROR   thonny.plugins.micropython.backend: Crash in backend
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 136, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 169, in _prepare_after_soft_reboot
    self._sys_path = self._fetch_sys_path()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 442, in _fetch_sys_path
    return self._evaluate("__thonny_helper.sys.path")
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 611, in _evaluate
    raise ManagementError(script, out, err)
thonny.plugins.micropython.backend.ManagementError: Problem with a management command

SCRIPT:
__thonny_helper.print_mgmt_value(__thonny_helper.sys.path)

STDOUT:

STDERR:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'path'

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

yubub, 2022-11-18:
I reinstall Thonny and now that what I get in the shell.

aivarannamaa, 2022-12-28:
Sorry for the late reply!

If you have `sys.py` on your MicroPython device, it may shadow the built-in module `sys`. What does your `sys.py` contain?

aivarannamaa, 2023-01-28:
I recommend renaming your sys.py and trying again.

closed 2023-01-28 by aivarannamaa