thonny/thonny #2528: SSL Certificate Issue

MOCS-BDobson, 2022-10-31:
To preface I am an IT Tech in a school:

**Collecting pygame
Could not fetch URL https://pypi.python.org/simple/pygame/: There was a problem confirming the ssl certificate: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:777) - skipping Could not find a version that satisfies the requirement pygame (from versions: ) No matching distribution found for pygame
Return code:**

Admins can install modules without issues. Teachers couldn't until I gave the group full control over Thonny program/data folders. Students cannot even with control over the same folders as Teachers. I messed with groups a bit and Students not hitting as restrictive a firewall can install modules. I spoke to the people that manage the firewall and they are insisting it's a Thonny issue, and that nothing is being blocked.

Does anyone have anything I can try?

aivarannamaa, 2022-10-31:
Does your issue look similar to #1986?

MOCS-BDobson, 2022-11-01:
> Does your issue look similar to #1986?

Definitely a close match, only real difference is that I'm using the inbuilt Python interpreter rather than MicroPython. The point about Chrome saying the license is the same is true as well as I can manually navigate to pypi.org and download the correct module (if I know which is correct), just not use the inbuilt package installer or pip. I'll try installing the certificate you provided and see if it sorts my issue. Thank you!

MOCS-BDobson, 2022-11-01:
> Does your issue look similar to #1986?

Do I need to basically get a certificate for pypi.org and get that on my test machine?

aivarannamaa, 2022-11-01:
> Do I need to basically get a certificate for pypi.org and get that on my test machine?
 
I suppose. 

I just found a page, which may help you: https://jhooq.com/pip-install-connection-error/

MOCS-BDobson, 2022-11-03:
> > Do I need to basically get a certificate for pypi.org and get that on my test machine?
> 
> I suppose.
> 
> I just found a page, which may help you: https://jhooq.com/pip-install-connection-error/

Step three on that one fixed it for me! Got a group policy creating the folder and ini file now for all students with Thonny and they can install packages just fine. Thank you so much for your help!

closed 2022-11-03 by MOCS-BDobson