thonny/thonny #2530: Exception while handling 'get_globals' (thonny.plugins.micropython.mp_back.ManagementError: Management markers missing).

ONE1CHINA, 2022-11-02:
The Problem occured suddenly after 2 weeks usage of the device.

Mac.OS
Info: >>> PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_globals' (thonny.plugins.micropython.mp_back.ManagementError: Management markers missing).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.

[backend.log](https://github.com/thonny/thonny/files/9921708/backend.log)

aivarannamaa, 2022-11-06:
Thank you for the report!

It looks like MicroPython crashed on your device:

> Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Which MicroPython version are you using? Perhaps a newer version fixes this?

ONE1CHINA, 2022-12-06:
Thank you for your reply! I do run the newest version of microPy. 
The problem has been solved: my main.py programm was wrong which was executing as the power is on....

closed 2022-12-06 by ONE1CHINA