thonny/thonny #2533: Support Russian

mmzhi, 2022-11-09:
Support Russian

SIMorLZS, 2022-11-09:
Support Russian!!!

aivarannamaa, 2022-11-10:
It's the first time someone demands support for a language in Thonny. So far people have expressed their interest more politely and I've directed them to https://github.com/thonny/thonny/blob/master/docs/translate.md, or they have found it themselves, implemented the translation and asked me to include it.

shreder95ua, 2022-11-16:
> Support Russian

rename this to "support naz*s"

Rodny0, 2022-11-22:
I have made a translation on POEditor.
Here is exported files:
[thonny_ru.zip](https://github.com/thonny/thonny/files/10067015/thonny_ru.zip)

aivarannamaa, 2022-11-22:
@Rodny0, thanks! I'll review the translation before the next release.

SIMself, 2022-11-22:
Many thanks for you. But I really felt sorry that I may have little time to check and give advice. 😭

我就是我
***@***.***

wildernessfamily, 2022-12-03:
Why support Russian? Russia is illegally invading and declaring war on the country that creates this amazing software. Instead, support Ukraine and these developers to help keep Russia from destroying this amazing nation!

sancarr, 2022-12-04:
@wildernessfamily Do you actually realize that the program is created by University of Tartu in Estonia. and not Ukraine?

shreder95ua, 2022-12-04:
@sancarr did he said about this university? this is not an argument. bruh

sancarr, 2022-12-04:
@shreder95ua I am having trouble understanding what you are trying to say. Your english is broken.

shreder95ua, 2022-12-05:
are you using translate or something? you're changing the theme, we're talking about different things.

aivarannamaa, 2022-12-05:
Let's keep the discussion focused. Russian translation seems to be completed in Poeditor. If you have comments about translation quality, then please use Poeditor's discussion facilities. Before the next release I'll pull the translation, make a review and the issue will be closed.

wildernessfamily, 2022-12-08:
@sancarr you may want to read what's on the main page of Thonny's GitHub https://github.com/thonny/thonny/wiki/Support-Ukraine
They don't even ask politely, they just state "Support Russian". Supporting Ukraine means not supporting Russia by allowing Russia to use a program like this to program their drones. As Thonny's GitHub page request, support Ukraine as Ukraine "must win" which means do not support Russia!

sancarr, 2022-12-08:
@wildernessfamily Bro.. This craziness and hatred has to stop.. You think Russian military doesn't have simple IDEs such as this to use for their own purposes? You are not stopping Russian military or anything with this type of attitude.. 

You don't understand one simple and fundamental fact that "Russia as a government" and "Russia as a people" are completely separate. Too many people do not understand this and thus all the hate going towards Russian people.. They are not to be hated. This war is not the people's fault. I hope you can make the simple distinction and have more understanding and tolerance towards "Russian people"

Tldr: stop merging the actions the government takes and its people into one whole "Russia" and hate all of it. Be able to see things calmly and clearly. You are a coder, are you not? Are you able to approach this subject as calmly and analytically as you do approach a new piece of code?

shreder95ua, 2022-12-08:
@sancarr if all russian peoples didn't continued this war, the war ended in 3 days, but Ukraine won. there are no "good russians".

sancarr, 2022-12-08:
@shreder95ua So much hatred without any logic... I wish peace upon your soul and I am done entertaining your ridiculous ideas.

aivarannamaa, 2022-12-12:
I understand (and feel) this hatred, but I still ask you to use this issue for discussing the quality of Russian translation.

For those who question the decision to include Russian translation in Thonny -- I believe that the Ukrainian theme of Thonny 4.0 is strong enough repellent for those Russian-speaking users who support Russia in this war. (Whether they are minority or majority of the potential Russian-speaking users, is the matter of another debate.)

wildernessfamily, 2022-12-19:
@sancarr that's why we have an unprovoked war on the amazing people of Ukraine and why we need to support Ukraine every possible way we can and NOT provide them with tools that they can use against Ukraine! Look at the videos from Ukraine of the amazing Ukrainian soldiers that take apart Russian drones. They are made with simple microcontrollers. Code that is used to write the instructions for those microcontrollers...Micro/Circuit-Python, C etc. IDE's like Thonny! But!, it's Russia as a whole that started this war. It's the people of Russia that has to condemn it and throw their leader out! That's why we sanction all of Russia! Not just the Kremlin! Common sense...ahhh, what an amazing thing when you use it...like grey matter! That's why THONNY IDE on the main website is asking to support Ukraine...and we MUST!!!

aivarannamaa, 2023-01-01:
Fixed in ba30fc2b

closed 2023-01-01 by aivarannamaa

Ivan228-s, 2023-02-02:
Hmm, and by chance no one noticed that the majority offends Russia, but a very small percentage offends Ukraine? I am a pure-blooded Ukrainian, I love Ukraine **And** Russia, I live in Russia, and I feel sorry for both sides, I am for peace. Support Russia **And** Ukraine!

Ivan228-s, 2023-02-02:
(p.s it doesn't matter in what order the sides with "and" are written, the sentences are equal)