thonny/thonny #2534: No political issue!

SIMself, 2022-11-09:
I don't want to support Ukraine, and I don't want to see political issues in a programming software!!!!!!

aivarannamaa, 2022-11-10:
In this case you have the option to not use Thonny.

closed 2022-11-10 by aivarannamaa

xkonglong, 2023-02-03:
> You'll always have the option to not use Thonny.

 I hope you can also support Syria refugees, Afghanistan refugees and all countries invaded by the United States. Thank you.

webdeb, 2023-06-10:
It's clear that the issue of helping Ukraine represents the opinion of the author, nevertheless how many people do share the same opinion it just sucks to provide your world views in software which should be neutral.

@xkonglong is right, the author shows his ignorance to other world conflict by not providing the same issue, like for the Ukraine conflict.

u07, 2023-10-02:
I've just updated and found that "support" button. Well, that is indeed unprofessional, to turn software into a propaganda tool. But as to the author's views, it's pretty much clear when it comes from. Estonia has been russophobic long before the current events (since WW2, probably). And that explains the biased attitude and ignorance to other world conflicts.

@aivarannamaa, I truly wish you had a broader vision on the world events. I have to stop using Thonny, but being honest with myself, I cannot but show you gratitude for sharing this convenient instrument with community, making the python ecosystem richer and more diverse. Have a good luck. -- Alex from Russia.