thonny/thonny #2536: Flashing Circuitpython to ESP32 without native USB

bborncr, 2022-11-11:
Starting in October, CircuitPython 8 supports ESP32 boards that don't appear as a Mass Storage Device. The Feather Huzzah32 is an example of this. Manually, the CircuitPython firmware can be installed using espytool at address `0x0000`.

Any suggestions on how/where flashing CircuitPython for these boards could be added?

Currently, the CircuitPython plugin only installs firmware using Mass Storage and the Micropython ESP32 flasher installs at `0x1000` and can't be changed as far as I can tell.

Thanks so much for Thonny!

aivarannamaa, 2022-11-12:
At the moment you can do this by selecting "MicroPython (ESP32)" interpreter and using its "Install MicroPython" tool. Simply select desired CircuitPython bin-file and start the flashing process.

davekeeshan, 2023-03-07:
I have hit this issue too with the LilyGo ESP32 C3.  I was able to run the commands that Thonny runs manually, but I had to change the base address:

    C:\downloads\thonny\python.exe -u -m esptool --port COM7 write_flash --flash_mode keep --flash_size detect 0x0 C:/downloads/esp32c3-20220117-v1.18.bin

Is it possible to change the base address via the `configuration.ini`?

FYI, the change is captured [here](https://micropython.org/download/esp32c3).

aivarannamaa, 2023-03-11:
I'll add the start-address parameter.

aivarannamaa, 2023-03-26:
Handled under #2707. Thonny 4.1 will have both UF2 and esptool installation links in CircuitPython config page.

closed 2023-03-26 by aivarannamaa