thonny/thonny #2538: Can't install any modules on Thonny

ThFrenchguy, 2022-11-12:
Hey (sorry for my bad English)  i manage to try many things with Tonny text editor, formy rRaspberry Pi Pico (i very like it btw exellent work !) but i have several issues when i tried to **install modules** (packages).
When I began the installation, this error line code appeared:

`ERROR: Can not combine '--user' and '--target'
Process returned with code 1`

But **i solved this** because my Rasp Pi was not plugged x')

**The problem now** is, when i start the installation, its collecting package but stop with an error 
`[end of output]
note: This error originates from a subprocess, and is likely not a problem with pip. error: subprocess-exited-with-error pip subprocess to install build dependencies did not run successfully. exit code: 1 See above for output. note: This error originates from a subprocess, and is likely not a problem with pip. Error Command '['C:\Users\33631\AppData\Local\pipkin\cache\workspaces\b1843a146278d117477687961e4e0504\Scripts\python.exe', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'micropython-ssd1306', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.`

I've seen about a package called **"Pipkin"** wich is a module who can download packages for Micropython, on PyPi, but i don't know how to use it. I tried many things, change the directory, install with the cmd and research with the Install from a local file, but it is still not working.

______________________________________________________________________________________________________________________________________

I use:

. Python version: 3.10
. Pip version: 22.3
. Thonny version: 4.0.1

Still working on it,  thank in advance :)

sebaguse, 2022-11-12:
Same here, can't install any module, very frustrating :( I found some suggestions about missing certificate which I installed but this didn't help unfortunately

aivarannamaa, 2022-11-14:
@ThFrenchguy, please copy the whole output, not only the end of it

SQTX, 2022-11-16:
Yeah, I've the similar situation. When I try install new package by packages menager i get error and this message:
```
Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [16/Nov/2022 02:21:32] "GET /gpio/ HTTP/1.1" 200 -
Collecting gpio
127.0.0.1 - - [16/Nov/2022 02:21:32] "GET /gpio/gpio-0.3.0.tar.gz/ HTTP/1.1" 200 -
  Downloading http://127.0.0.1:36628/gpio/gpio-0.3.0.tar.gz/
     - 2.9 kB ? 0:00:00
  Installing build dependencies: started
127.0.0.1 - - [16/Nov/2022 02:21:34] "GET /setuptools/ HTTP/1.1" 200 -
127.0.0.1 - - [16/Nov/2022 02:21:34] "GET /setuptools/setuptools-65.5.1-py3-none-any.whl/ HTTP/1.1" 200 -
127.0.0.1 - - [16/Nov/2022 02:21:35] "GET /wheel/ HTTP/1.1" 200 -
127.0.0.1 - - [16/Nov/2022 02:21:35] "GET /wheel/wheel-0.38.4-py3-none-any.whl/ HTTP/1.1" 200 -
127.0.0.1 - - [16/Nov/2022 02:21:37] "GET /pip/ HTTP/1.1" 200 -
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Building wheels for collected packages: gpio
  Building wheel for gpio (pyproject.toml): started
  Building wheel for gpio (pyproject.toml): finished with status 'done'
  Created wheel for gpio: filename=gpio-0.3.0-py3-none-any.whl size=3211 sha256=392330bb9c89b9e9e307366a4be7be5a62c5b340900fddb50f1fe56e21ac26ce
  Stored in directory: /Users/sqtx/Library/Caches/pipkin/wheels/af/85/a4/374fd1b812d70c2baec51a65df15ce2fe8f15aad440cd93ff1
Successfully built gpio
Installing collected packages: gpio
Successfully installed gpio-0.3.0
Starting to apply changes to the target.
Copying gpioError (ManagementError(...), 'Command output was not empty')
```

After closing the install window, I see this text in the right bottom:
[image]
How can I fix this?

I use :
- Thonny 4.0.1
- Python 3.10.8
- Pip 22.2.2

Thank in advance.

ThFrenchguy, 2022-11-18:
Hey again @aivarannamaa , and sorry for the lack of informations, here is the error code i get : 

`Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [18/Nov/2022 22:58:30] "GET /micropython-ssd1306/ HTTP/1.1" 200 -
Collecting micropython-ssd1306
  Using cached http://127.0.0.1:36628/micropython-ssd1306/micropython-ssd1306-0.3.tar.gz/
  Installing build dependencies: started
127.0.0.1 - - [18/Nov/2022 22:58:34] "GET /setuptools/ HTTP/1.1" 200 -
127.0.0.1 - - [18/Nov/2022 22:58:38] "GET /wheel/ HTTP/1.1" 200 -
  Installing build dependencies: finished with status 'error'
  error: subprocess-exited-with-error
  
  pip subprocess to install build dependencies did not run successfully.
  exit code: 1
  
  [10 lines of output]
  Looking in indexes: http://127.0.0.1:36628
  Collecting setuptools>=40.8.0
    Using cached http://127.0.0.1:36628/setuptools/setuptools-65.5.1-py3-none-any.whl/
  Collecting wheel
    Using cached http://127.0.0.1:36628/wheel/wheel-0.38.4-py3-none-any.whl/
  Installing collected packages: wheel, setuptools
  ERROR: Cannot set --home and --prefix together
  
  [notice] A new release of pip available: 22.2.1 -> 22.3.1
  [notice] To update, run: C:\Users\33631\AppData\Local\pipkin\cache\workspaces\b1843a146278d117477687961e4e0504\Scripts\python.exe -m pip install --upgrade pip`

I also use Win 11.

mbradley612, 2022-11-26:
I'm also seeing this error. Apple M1 with Ventura 13.0

About Thonny reports:

Thonny 4.0
Darwin 22.1.0
Python 3.10.6
Tk 8.6.12

Some packages install OK, e.g. micropython_unittest

When I try to install micropython-umqtt.simple, I get the following:

```
Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [26/Nov/2022 00:27:37] "GET /micropython-umqtt-robust2/ HTTP/1.1" 200 -
Collecting micropython-umqtt.robust2
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 54436)
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 204, in __init__
    super().__init__(request, client_address, server)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 432, in handle
    self.handle_one_request()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 420, in handle_one_request
    method()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 211, in do_GET
    self._serve_file(*path.split("/"))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 245, in _serve_file
    tweaked_bytes = self._tweak_file(dist_name, file_name, original_bytes)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 312, in _tweak_file
    assert normalize_dist_name(wrapper_dir).startswith(normalize_dist_name(dist_name))
AssertionError
----------------------------------------
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /micropython-umqtt-robust2/micropython-umqtt.robust2-2.1.1.tar.gz/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 54438)
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 204, in __init__
    super().__init__(request, client_address, server)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 432, in handle
    self.handle_one_request()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/http/server.py", line 420, in handle_one_request
    method()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 211, in do_GET
    self._serve_file(*path.split("/"))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 245, in _serve_file
    tweaked_bytes = self._tweak_file(dist_name, file_name, original_bytes)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/vendored_libs/pipkin/proxy.py", line 312, in _tweak_file
    assert normalize_dist_name(wrapper_dir).startswith(normalize_dist_name(dist_name))
AssertionError
----------------------------------------
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /micropython-umqtt-robust2/micropython-umqtt.robust2-2.1.1.ta [... cut ...] lError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response')))

Error Command '['/Users/mbradley/Library/Caches/pipkin/workspaces/d64e98649188e5065d1f585e6f96486a/bin/python3', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'micropython-umqtt.robust2', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.

```

magnussommarsjo, 2022-11-28:
When downgrading to `thonny 3.3.14` installing packages works again.  
I have tried and got this issue both in `thonny 4.0.0` and `thonny 4.0.1`

TszYiYang, 2022-12-10:
@magnussommarsjo I took your approach and it worked! I revert to thonny ver.3.3.13 and the package downloads smoothly. 

The system I used when having download package issue:

macOS Ventura 13.0.1
Thonny 4.0.1
Python 3.10.8

ghazanhaider, 2022-12-15:
Had the same errors in 4.0.1, the same workaround worked.. downgraded to 3.3.13.
4.0.1 also did not list the existing installed modules.

Windows 10
Raspberry PI Pico W

plopp, 2022-12-18:
I'm experiencing the exact same error for `micropython-umqtt.robust` and Thonny version 4.0.1.

aivarannamaa, 2022-12-28:
The errors when installing a MicroPython package with dot in its name (https://github.com/thonny/thonny/issues/2524#issuecomment-1366792073) are different from the original problem in this thread.

@ThFrenchguy, sorry for the late reply! It looks like your `pip` isn't compatible with `wheel` package, but I don't understand how this can be as both `pip` and `wheel` versions are fixed in pipkin. I have some pipkin changes coming for Thonny 4.0.2 and I hope they will fix your problem as well.

MisterSoandSo, 2022-12-30:
So it seems like the temporary fix is to downgrade Thonny.

MisterSoandSo, 2022-12-30:
Nevermind, you can still manually download the library from their respective repositories.

arjjung29, 2023-01-03:
hey @aivarannamaa using thonny 4.0.1 and still not able to install umqtt.simple via tools>manage packages

i tried extracting the zip file and then adding the umqtt.simple.py file mannually still getting 

MPY: soft reboot
network config: ('192.168.0.114', '255.255.255.0', '192.168.0.1', '202.168.158.245')
1014.18 hPa
Traceback (most recent call last):
  File "main.py", line 22, in <module>
  File "umqttsimple.py", line 112, in publish
TypeError: object of type 'bound_method' has no len()
MicroPython v1.19.1-782-g699477d12 on 2022-12-20; Raspberry Pi Pico W with RP2040
Type "help()" for more information.

my main.py file ....

# PiicoDev MS5637 minimal example code
# This program temperature and pressure data from the PiicoDev MS5637 pressure sensor
# and displays the result

from umqttsimple import MQTTClient
from machine import Pin
from time import sleep
from PiicoDev_MS5637 import PiicoDev_MS5637
from PiicoDev_Unified import sleep_ms

pressure = PiicoDev_MS5637()
CLIENT_NAME =b'pressure sensor'
BROKER_ADDR = b'192.168.0.111'
mqttc = MQTTClient(CLIENT_NAME, BROKER_ADDR, keepalive=60)

while True:
    press_hPa = pressure.read_pressure()
    altitude_m = pressure.read_altitude()
    a = (str(press_hPa)).encode
    # Print Pressure
    print(str(press_hPa) + " hPa")
    mqttc.publish( b'pressure sensor', a)
    
    # Print Altitude (metres)
#     print(str(altitude_m) + " m")
    sleep_ms(100)

my umqttsimple.py file ......

import usocket as socket
import ustruct as struct
from ubinascii import hexlify

class MQTTException(Exception):
    pass

class MQTTClient:

    def __init__(self, client_id, server, port=0, user=None, password=None, keepalive=0,
                 ssl=False, ssl_params={}):
        if port == 0:
            port = 8883 if ssl else 1883
        self.client_id = client_id
        self.sock = None
        self.server = server
        self.port = port
        self.ssl = ssl
        self.ssl_params = ssl_params
        self.pid = 0
        self.cb = None
        self.user = user
        self.pswd = password
        self.keepalive = keepalive
        self.lw_topic = None
[... 156 lines cut ...]
        topic_len = self.sock.read(2)
        topic_len = (topic_len[0] << 8) | topic_len[1]
        topic = self.sock.read(topic_len)
        sz -= topic_len + 2
        if op & 6:
            pid = self.sock.read(2)
            pid = pid[0] << 8 | pid[1]
            sz -= 2
        msg = self.sock.read(sz)
        self.cb(topic, msg)
        if op & 6 == 2:
            pkt = bytearray(b"\x40\x02\0\0")
            struct.pack_into("!H", pkt, 2, pid)
            self.sock.write(pkt)
        elif op & 6 == 4:
            assert 0

    # Checks whether a pending message from server is available.
    # If not, returns immediately with None. Otherwise, does
    # the same processing as wait_msg.
    def check_msg(self):
        self.sock.setblocking(False)
        return self.wait_msg()

Can someone please help me !!!!!!!

magnussommarsjo, 2023-01-03:
@arjjung29 try downgrading thonny to sub 4.0.0. Then install via package manager.

> When downgrading to `thonny 3.3.14` installing packages works again.  
> I have tried and got this issue both in `thonny 4.0.0` and `thonny 4.0.1`

arjjung29, 2023-01-03:
Yea thanks for the tip when i did it said i missed python39.dll file i got frustated i spent almost 1.30 hours resolving it, then i imported the umqtt.simple.py code manually and still got error then i found out where i commited my mistake...... whcih is that i did not implement **mqttc.connect()** so why it showed me a network error and i didn't enocde the message while publishing using the **mqttc.publish()** finally i fixed it and got my code working. Thanks for the tip @magnussommarsjo

aivarannamaa, 2023-01-16:
https://github.com/thonny/thonny/releases/tag/v4.0.2 is now out. Please reopen if you still have the same problem.

closed 2023-01-16 by aivarannamaa

magnussommarsjo, 2023-01-17:
@aivarannamaa  Installing packages in `4.0.2` works for me now. Thanks!

falconsol, 2023-01-18:
I have been experiencing the same problem installing packages, specifically umqtt-simple.

Here is the output of the error:

install --user micropython-umqtt.simple
Collecting micropython-umqtt.simple
  Using cached micropython-umqtt.simple-1.3.4.tar.gz (4.0 kB)
ERROR: micropython-umqtt.simple from https://files.pythonhosted.org/packages/bd/cf/697e3418b2f44222b3e848078b1e33ee76aedca9b6c2430ca1b1aec1ce1d/micropython-umqtt.simple-1.3.4.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Process returned with code 1

falconsol, 2023-01-18:
I also am experiencing the same problem installing packages, specifically the **micropython-umqtt.simple** package.

I just installed Thonny 4.0.2 as suggest with the same result.

I have tried Windows 4.0.1
Linux 4.0.1
and now Windows 4.0.2 all with the same error.

install --user micropython-umqtt.simple
Collecting micropython-umqtt.simple
  Using cached micropython-umqtt.simple-1.3.4.tar.gz (4.0 kB)
ERROR: micropython-umqtt.simple from https://files.pythonhosted.org/packages/bd/cf/697e3418b2f44222b3e848078b1e33ee76aedca9b6c2430ca1b1aec1ce1d/micropython-umqtt.simple-1.3.4.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Process returned with code 1

aivarannamaa, 2023-01-18:
@falconsol, did you use "Tools => Manage packages" or "Tools => Manage plug-ins"

falconsol, 2023-01-18:
I tried both ways. Both fail the same way.

aivarannamaa, 2023-01-19:
@falconsol, what does it say in the lower-right corner of Thonny's main window when you are going to install the package?

collinr, 2023-02-07:
> I also am experiencing the same problem installing packages, specifically the **micropython-umqtt.simple** package.
>
> I just installed Thonny 4.0.2 as suggest with the same result.
>
> I have tried [...]

you need to change to "MicroPython (Raspberry Pi Pico)" by clicking at the bottom right where it says "Python"

zcool74, 2023-02-11:
> you need to change to "MicroPython (Raspberry Pi Pico)" by clicking at the bottom right where it says "Python"

Perfect solution,
Thx

falconsol, 2023-02-12:
Thank you! That worked!

Thiago-FR, 2023-04-14:
I also am experiencing the same problem installing packages. Thonny 4.0.2

```
install micropython-uasyncio
Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [14/Apr/2023 11:10:15] "GET /micropython-uasyncio/ HTTP/1.1" 200 -
Collecting micropython-uasyncio
  Using cached micropython_uasyncio-3.1.1-py3-none-any.whl
127.0.0.1 - - [14/Apr/2023 11:10:16] "GET /micropython-uasyncio-core/ HTTP/1.1" 200 -
Collecting micropython-uasyncio.core
  Using cached micropython_uasyncio.core-2.3-py3-none-any.whl
Installing collected packages: micropython-uasyncio.core, micropython-uasyncio
Successfully installed micropython-uasyncio-3.1.1 micropython-uasyncio.core-2.3
Starting to apply changes to the target.
Copying micropython_uasyncio.coreError (ManagementError(...), 'Command output was not empty')
```

```
install install /home/thiago/Downloads/micropython-uasyncio-3.1.1.tar.gz
Looking in indexes: http://127.0.0.1:36628
Processing ./Downloads/micropython-uasyncio-3.1.1.tar.gz
ERROR: file:///home/thiago/Downloads/micropython-uasyncio-3.1.1.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Error Command '['/home/thiago/.cache/pipkin/workspaces/77b23d5cc84289b84f594d790c14a557/bin/python3', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'install', '/home/thiago/Downloads/micropython-uasyncio-3.1.1.tar.gz', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.
```

aivarannamaa, 2023-04-15:
@Thiago-FR, do you have enough space on your device's flash?

micheld93, 2023-07-09:
I have a similar problem with version 4.1.1. I get this message: 

`/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10 -m pip --disable-pip-version-check --no-color install --no-warn-script-location --user --user micropython-bme280
Collecting micropython-bme280
  Using cached micropython-bme280-2.1.3.tar.gz (3.3 kB)
ERROR: micropython-bme280 from https://files.pythonhosted.org/packages/e4/dc/549e2aaafb3ba9a38f304b809c4bdaa12b1d76979698c038db757740d31b/micropython-bme280-2.1.3.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Error: process returned with code 1
`

What do I need to get this module? Other modules can be downloaded.

micheld93, 2023-10-21:
The problem above is still existing in version 4.1.3. I use a Mac with Ventura installed. There is enough space on my machine. Even the work around mentioned in https://github.com/thonny/thonny/issues/2538#issuecomment-1426854836 does not help. 

Any further ideas?

Ransu242, 2024-03-30:
someong help me. im trying to install pyrebase for my rpi pico w using thonny ide i can't install pyrebase package it said it doesnt look like a micropython package and if i install it an error occur

[image]

Installing to temp directory
"C:\Program Files (x86)\Thonny\python.exe" -m thonny.plugins.micropython.minipip install --target C:\Users\USER\AppData\Local\Temp\tmpbm4c2jqz Pyrebase
Processing 'Pyrebase'
Querying package metadata from https://micropython.org/pi/Pyrebase/json
Could not find 'Pyrebase' from https://micropython.org/pi
Querying package metadata from https://pypi.org/pypi/Pyrebase/json
Inspecting version 3.0.27
'Pyrebase' will be installed with pip (not having single tar.gz asset).
Installing with pip: ['Pyrebase']
Collecting Pyrebase
  Using cached Pyrebase-3.0.27-py3-none-any.whl.metadata (681 bytes)
Collecting gcloud==0.17.0 (from Pyrebase)
  Using cached gcloud-0.17.0-py3-none-any.whl
Collecting oauth2client==3.0.0 (from Pyrebase)
  Using cached oauth2client-3.0.0-py3-none-any.whl
Collecting pycryptodome==3.4.3 (from Pyrebase)
  Using cached pycryptodome-3.4.3.tar.gz (6.5 MB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting python-jwt==2.0.1 (from Pyrebase)
  Using cached python_jwt-2.0.1-py2.py3-none-any.whl.metadata (7.5 kB)
Collecting requests==2.11.1 (from Pyrebase)
  Downloading requests-2.11.1-py2.py3-none-any.whl.metadata (39 kB)
Collecting requests-toolbelt==0.7.0 (from Pyrebase)
  Using cached requests_toolbelt-0.7.0-py2.py3-none-any.whl.metadata (11 kB)
Collecting httplib2>=0.9.1 (from gcloud==0.17.0->Pyrebase)
  Using cached httplib2-0.22.0-py3-none-any.whl.metadata (2.6 kB)
Collecting googleapis-common-protos (from gcloud==0.17.0->Pyrebase)
  Using cached googleapis_common_protos-1.63.0-py2.py3-none-any.whl.metadata (1.5 kB)
Collecting protobuf!=3.0.0.b2.post1,>=3.0.0b2 (from gcloud==0.17.0->Pyrebase)
  Using cached protobuf-4.24.4-cp37-cp37m-win32.whl.metadata (540 bytes)
Collecting six (from gcloud==0.17.0->Pyrebase)
  Using cached six-1.16.0-py2.py3-none-any.whl.metadata (1.8 kB)
Collecting pyasn1>=0.1.7 (from oauth2client==3.0.0->Pyrebase)
  Using cached pyasn1-0.5.1-py2.py3-none-any.whl.metadata (8.6 kB)
Collecting pyasn1-modules>=0.0.5 (from oauth2client==3.0.0->Pyrebase)
  Using cached pyasn1_modules-0.3.0-py2.py3-none-any.whl.metadata (3.6 kB)
Collecting rsa>=3.1.4 (from oauth2client==3.0.0->Pyrebase)
  Using cached rsa-4.9-py3-none-any.whl.metadata (4.2 kB)
Collecting jws>=0.1.3 (from python-jwt==2.0.1->Pyrebase)
  Using cached jws-0.1.3-py3-none-any.whl
Collecting pyparsing!=3.0.0,!=3.0.1,!=3.0.2,!=3.0.3,<4,>=2.4.2 (from httplib2>=0.9.1->gcloud==0.17.0->Pyrebase)
  Using cached pyparsing-3.1.2-py3-none-any.whl.metadata (5.1 kB)
Using cached Pyrebase-3.0.27-py3-none-any.whl (9.6 kB)
Using cached python_jwt-2.0.1-py2.py3-none-any.whl (8.8 kB)
Downloading requests-2.11.1-py2.py3-none-any.whl (514 kB)
   ---------------------------------------- 514.8/514.8 kB 2.5 MB/s eta 0:00:00
Using cached requests_toolbelt-0.7.0-py2.py3-none-any.whl (52 kB)
Using cached httplib2-0.22.0-py3-none-any.whl (96 kB)
Using cached protobuf-4.24.4-cp37-cp37m-win32.whl (409 kB)
Using cached pyasn1-0.5.1-py2.py3-none-any.whl (84 kB)
Using cached pyasn1_modules-0.3.0-py2.py3-none-any.whl (181 kB)
Using cached rsa-4.9-py3-none-any.whl (34 kB)
Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Using cached googleapis_common_protos-1.63.0-py2.py3-none-any.whl (229 kB)
Using cached pyparsing-3.1.2-py3-none-any.whl (103 kB)
Building wheels for collected packages: pycryptodome
  Building wheel for pycryptodome (setup.py): started
  Building wheel for pycryptodome (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  python setup.py bdist_wheel did not run successfully.
  exit code: 1
  
  [364 lines of output]
  running bdist_wheel
[... 349 lines cut ...]
  copying lib\Crypto\SelfTest\Signature\test_vectors\ECDSA\SigVer.rsp -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\ECDSA
  creating build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
  copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5\SigGen15_186-2.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
  copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5\SigGen15_186-3.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
  copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5\SigVer15_186-3.rsp -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
  creating build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
  copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS\SigGenPSS_186-2.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
  copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS\SigGenPSS_186-3.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
  copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS\SigVerPSS_186-3.rsp -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
  Skipping optional fixer: buffer
  Skipping optional fixer: idioms
  Skipping optional fixer: set_literal
  Skipping optional fixer: ws_comma
  running build_ext
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycryptodome
  Running setup.py clean for pycryptodome
Failed to build pycryptodome
ERROR: Could not build wheels for pycryptodome, which is required to install pyproject.toml-based projects

minipip returned with error code 1

[image]