thonny/thonny #2542: cannot install any packages

MoonlightFlora, 2022-11-16:
I can't install any packages.
When I try to search up pygame (the package I want) nothing shows up. There is only an endless loading button
[image]

aivarannamaa, 2022-11-19:
Thank you for the report!

What happens when you select "Tools => Open system shell" and enter `pip install --user pygame`?

aivarannamaa, 2022-12-28:
@kyubird, are you trying to run this script on a MicroPython device? Most of the packages containing setup.py are meant for CPython and there is no chance it could work with MicroPython.

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and have more information to add.

closed 2023-01-28 by aivarannamaa