thonny/thonny #2543: CTRL-C Exit Button

anwarbashir, 2022-11-16:
Hi,
I am demonstrating programs using Thonny (v 3.3.14) on Raspberry Pi. This includes robotics and motors. In Python3 there is a try Exception for keyboard interrupt and I can use this to complete a GPIO.cleanup(), ensuring that the pins are not left high. I am finding that the Stop/Restart button performs a different function and will leave the pins high. I can not find a way to capture the code and execute the GPIO.cleanup(). My users are beginners so they will automatically try to press the Stop Button, assuming that it will stop the program as it would in most USE cases. So is there a way to add/optional add another button that performs the same function as CTRL-C.

aivarannamaa, 2022-11-16:
Hi!

I heard that RPi will soon get Thonny 4.0.1 with slightly different Stop/Restart behavior. In 3.3 this button was overly eager by sending several Ctrl-C-s in a row so that the second one could have interrupted the code in the `except KeyboardInterrupt:` block.

Thonny 4.0 is supposed to send a Ctrl-C and wait for 100ms before sending the next one (in case the first one didn't give it the REPL).

aivarannamaa, 2023-01-28:
RPi now has the honny 4.0

closed 2023-01-28 by aivarannamaa