thonny/thonny #2544: Thonny3.3.13 Can't run multiprocessing program, Error: AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>

ryle2020, 2022-11-17:
Thonny: 3.3.13
OS: Windows 10
Program:
```python
import time, random
from multiprocessing import Process

data = []
for i in range(10000):
    rand = random.randint(20000, 100000000)
    data.append(rand)

def calPrime(n):
    primfac = []
    d = 2
    while d * d <= n:
        while (n % d) == 0:
            primfac.append(d)
            n //= d
        d += 1
    if n > 1:
        primfac.append(n)
    return primfac

def executeProc(n):
    for i in range(n * 1000, (n + 1) * 1000):
        print(f"R2-Process{n}: {calPrime(i)}")

def main():
    print("Starting number crunching")

    # method one : single
    t0 = time.time()
    for num in data:
        print(f"R1: {calPrime(num)}")
    t1 = time.time()

    print("#" * 30)
    print("#" * 30)
    
    # method two: multiprocessing
    t2 = time.time()
    procs = []
    for i in range(10):
        proc = Process(target=executeProc, args=(i,))
        procs.append(proc)
        proc.start()
    for proc in procs:
        proc.join()
    t3 = time.time()

    print(f"Execution Time for result1:{t1-t0}")
    print(f"Execution Time for result2:{t3-t2}")

if __name__ == "__main__":
    main()

```

The program could be run normally  in CMD or in MU(another editor), but could not be run successfully in Thonny. 
The error message:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
Traceback (most recent call last):
  File "<string>", line 1, in <module>
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Users\XXX\.pyenv\pyenv-win\versions\3.9.13\lib\multiprocessing\spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
AttributeError: Can't get attribute 'executeProc' on <module '__main__' (built-in)>

aivarannamaa, 2022-11-18:
Please try Thonny 4.0.1

ryle2020, 2022-11-18:
tested in linux(linux-mint: Linux Mint 20.3， Kernel: Linux 5.4.0-125-generic) 
The program is run normally

closed 2023-01-28 by aivarannamaa