thonny/thonny #2547: How does this project package my modified code into an exe file?

rahmanmy, 2022-11-21:
I have added some functions to this project. I think I will change this by myself and package it into a complete running software. How to pack it? 
Now the packing is still the same, not my own change, what to do? 
Wait on the line!

aivarannamaa, 2022-11-25:
Do I undestand properly that you want to package a modified version of Thonny? You may want to explore the contents of https://github.com/thonny/thonny/tree/master/packaging

rahmanmy, 2022-11-26:
I already explore the contents of https://github.com/thonny/thonny/tree/master/packaging .I running this file https://github.com/thonny/thonny/blob/master/packaging/windows/create_installer_38.bat.
I have run this file to package a new version of thonny. After packaging, I ran thonny.exe. I found that the function I added was still unavailable. It is the same as the official download of thonny.exe. There is no version I want. Can you tell me how to package your modified code into a new version of thonny。thank you!

aivarannamaa, 2022-12-28:
I don't know what and how did you add, so I'm afraid you need to debug it yourself.

closed 2022-12-28 by aivarannamaa