thonny/thonny #2551: [feature request] Show file properties by right clicking on the tab

smoore-33, 2022-11-23:
I'm just starting with Thonny and Micropython. I find it difficult to remember whether versions of open modules shown on the tabs are from the Pico and which are from the folder on my laptop, where I keep copies of the code. While there may be a better way of keeping track of this, I would suggest that in any case, it would be helpful to be able to right-click on each tab (where Close, Close others, and Close all appear), then click a Properties option to show the file location and other file properties. Or perhaps there already is a better way to do this that I am unaware of.

aivarannamaa, 2022-11-25:
When you click on an editor tab, then the whole path is shown at the titlebar of Thonny's main window. For non-local files it also shows the kind of the location.

In case you simply want to distinguish between local and non-local files -- non-local files (including the files on your Pico) have square brackets around their name at tab title.

smoore-33, 2022-11-25:
Thanks for the help. That takes care of it.

closed 2022-11-25 by smoore-33