thonny/thonny #2552: AttributeError: no such attribute

jouellnyc, 2022-11-23:
Hello. It appears I've triggered an issue with Thonny 4.0:

Hopefully helpful background:

I just flashed a D1 Mini fine with Thonny 4.0 (also tried w/4.1) (w/'dio' mode) OK but cannnot get a REPL post reboot:

[image]

- I can connect with gtkterm OK to get a REPL and I was able to download Thonny 3.X (I just grabbed an earlier version randomly) and get a REPL as well.

- I have an esp32 connected via the other USB port so **it seems like** maybe this is D1 mini or esp8266 related...

Here's the  backend.log

```
backend.log
21:34:22.443 INFO    thonny: Thonny version: 4.0.1
21:34:22.444 INFO    thonny: cwd: /home/john
21:34:22.444 INFO    thonny: original argv: ['/home/john/apps/thonny/bin/python3.10', '-u', '-B', '/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
21:34:22.444 INFO    thonny: sys.executable: /home/john/apps/thonny/bin/python3.10
21:34:22.444 INFO    thonny: sys.argv: ['/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp/esp_back.py', "{'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
21:34:22.444 INFO    thonny: sys.path: ['/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/esp', '/home/john/apps/thonny/lib/python310.zip', '/home/john/apps/thonny/lib/python3.10', '/home/john/apps/thonny/lib/python3.10/lib-dynload', '/home/john/.local/lib/python3.10/site-packages', '/home/john/apps/thonny/lib/python3.10/site-packages']
21:34:22.444 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
21:34:22.444 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/ttyUSB0', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': False, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP8266Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
21:34:22.445 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
21:34:22.445 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
21:34:22.445 INFO    thonny: TIME/MODS 0.035 before prepare (+73 modules)
21:34:22.445 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
21:34:22.446 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
21:34:22.446 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
21:34:22.464 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
21:34:22.474 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
21:34:22.479 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'OK\r\n'
21:34:22.479 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'MPY: soft reboot\r\n'
21:34:22.506 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
21:34:22.530 INFO    thonny: TIME/MODS 0.085 got welcome 
21:34:22.530 INFO    thonny: TIME/MODS 0.000 bef preparing helpers 
21:34:22.530 INFO    thonny.plugins.micropython.mp_back: Preparing helpers
21:34:22.530 INFO    thonny: TIME/MODS 0.000 befsubcode 
21:34:22.535 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
21:34:22.545 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
21:34:22.680 INFO    thonny: TIME/MODS 0.150 affsubcode 
21:34:22.707 INFO    thonny: TIME/MODS 0.027 affforw 
21:34:22.707 INFO    thonny: TIME/MODS 0.000 prepared helpers 
21:34:22.707 INFO    thonny: TIME/MODS 0.000 befsubcode 
21:34:22.718 INFO    thonny: TIME/MODS 0.011 affsubcode 
21:34:22.744 INFO    thonny: TIME/MODS 0.026 affforw 
21:34:22.744 ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 151, in __init__
    self._prepare_after_soft_reboot(clean)
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 185, in _prepare_after_soft_reboot
    self._update_cwd()
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 458, in _update_cwd
    super()._update_cwd()
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 449, in _update_cwd
    self._cwd = self._evaluate("__thonny_helper.getcwd()")
  File "/home/john/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 571, in _evaluate
    raise ManagementError("Script produced errors", script, out, err)
thonny.plugins.micropython.mp_back.ManagementError: Script produced errors
21:34:22.744 ERROR   thonny.plugins.micropython.mp_back: ManagementError details:
SCRIPT: __thonny_helper.print_mgmt_value(__thonny_helper.getcwd())

STDOUT: 

STDERR: Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 42, in getcwd
AttributeError: no such attribute

```

This is on Ubuntu 22.04.1 LTS

(Thank you for a great IDE BTW, nothing else come close IMHO)

aivarannamaa, 2022-11-25:
Thank you for the report!

When you get the REPL with gtkterm, what happens when you execute following commands:

```python
import uos
uos.getcwd()
import os
os.getcwd()
```

?

jouellnyc, 2022-11-26:
Sure. Thing! Here's the output as requested:

```
MicroPython v1.19.1-705-gac5934c96 on 2022-11-18; ESP module (512K) with ESP8266
Type "help()" for more information.
>>> 
>>> import uos
>>> uos.getcwd()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: no such attribute
>>> import os
>>> os.getcwd()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: no such attribute
>>> 

```

jouellnyc, 2022-11-26:
Also. your comment made me ask myself a question. Which FW did use. 

At first (above) I used "ESP8266 with 512kiB flash" version, specifcally esp8266-512k-20221118-unstable-v1.19.1-705-gac5934c96.bin  from https://micropython.org/download/esp8266-512k/

Then I tried the "ESP8266 with 2MiB+ flash" version:
https://micropython.org/resources/firmware/esp8266-20220618-v1.19.1.bin

This works and Thonny is happy. So now that's resolved I guess I will leave it up to you if there's any value in this information. Of course if I started w/the 2MB version, I would never had had this problem =D.

aivarannamaa, 2022-11-26:
Thanks, this is good to know, at least for explaining next issues. I'm surprised that the "ESP8266 with 512kiB flash" version did not have the `getwcd` function, as if it didn't have any filesystem at all.

jouellnyc, 2022-11-26:
Right. I figured it could be good to know. Sometimes mistakes (mine here) can be fruitful!

closed 2022-12-28 by aivarannamaa