thonny/thonny #2553: Thonny 4.0.1 'Manage packages' option greys out

Bold-engi, 2022-11-23:
Hi, I'm using thonny to programme my raspberry pi board to control a stepper motor. I want to do it by using RPi.GPIO but I can't open the 'Manage packages' to install the package. I tried using 'Manage plug-in' options but it just stuck on the installation process and never get to finish.

This below is the greys out 'Manage package' screenshot:
[image]

This below is the 'never-ending' installation screenshot:
[image]

Please don't mind the red text below, that's because I didn't connect any of my board to my laptop, it shouldn't be the reason why I can't get the package installed.

Many thanks

aivarannamaa, 2022-11-24:
> Please don't mind the red text below, that's because I didn't connect any of my board to my laptop, it shouldn't be the reason why I can't get the package installed.

Well, this is exactly the reason you can't get the package installed, because the packages are supposed to end up on the board. Or do you want to install a Python 3 package? In this case you need to switch to Python 3 mode.

Bold-engi, 2022-11-24:
Hi @aivarannamaa , thanks for poining out where my mistake is. It does because I didn't plug my board on. The problem solved after I make connection.

closed 2022-11-25 by aivarannamaa