thonny/thonny #2557: ESP32 and Thonny, Device is busy or does not respond,Could not interrupt current process. Please wait, try again or select Stop/Restart!

ExcaliburAvalon, 2022-11-25:
When I use arduino, I can program and open ports, and when I use thonny, I can't save the program to esp32, and it replies to me with this message
Device is busy or does not respond. Your options:

wait until it completes current work;
use Ctrl+C to interrupt current work;
reset the device and try again;
check connection properties;
make sure the device has suitable MicroPython / CircuitPython / firmware;
make sure the device is not in bootloader mode.
I tried the following method, but nothing happened.
Click on Reset button
Stop/Restart Backend OR Ctrl+F2
[image]

ExcaliburAvalon, 2022-11-28:
Thonny ---Run---configure interpreter---interpreter---Install or update MicroPython
Choose Port and Firmware,Insatall.
When the download is complete,it's OK.

aivarannamaa, 2022-12-28:
I assume you have found solution to your problem. Please reopen if problem persists.

SaurabhDwivedii, 2023-01-22:
I am using ESP8266 and getting same error , tried your above solution but not working.
please help

aivarannamaa, 2023-01-26:
@SaurabhDwivedii, can you get your MicroPython REPL with other means (Putty, tio, screen, ...) ?

closed 2023-01-28 by aivarannamaa

SaurabhDwivedii, 2023-01-29:
yes, when i tried putty that time not able to connect

SaurabhDwivedii, 2023-01-29:
> 

@aivarannamaa  yes, when i tried with putty that time not able to connect

aivarannamaa, 2023-01-29:
In this case the you should seek help at https://github.com/orgs/micropython/discussions

rippershinoda, 2023-09-26:
Hola a todos, tenia el mismo problema con un ESP32 CAM, buscando en otros foros ya nos habían otorgado la solución
Solución: 

Hay que editar el archivo confi.ini de Thonny.

Y en la parte de [ESP32]
hay que agregar estas lineas:

dtr = False
rts = False

El archivo config.ini se edita desde Tools , Open Thonny Data Folder
Se debe editar el archivo desde antes de que se intente instalar el firmware.
Y listo, se soluciona el problema.

desarrollador1SMI, 2023-10-26:
> Hola a todos, tenia el mismo problema con un ESP32 CAM, buscando en otros foros ya nos habían otorgado la solución Solución:
>
> Hay que editar el archivo confi.ini de Thonny.
>
> Y en la parte de [ESP32] [...]

me ha funcionado esta opción. mil gracias

Maniprogramer, 2024-03-15:
First download  micro-python flasher file you can find this flasher file in this link "http://www.micropython.org/download/" after you visit this page proceed as per your board accordingly and download any one bin file of last 4-5 latest version of your choice.

Now you need to upload this flasher file to the board that your are working with in order to flash this bin file you need a flasher app to make this process work and you can find this flasher app for windows "https://github.com/nodemcu/nodemcu-flasher" 

for mac and linux users you can find in "https://github.com/marcelstoer/nodemcu-pyflasher" and then try to upload micro python bin file to the board in my case i'am using esp8266 after done uploading the flasher file to the board 

now you can try to upload the micro python code that you've written in thonny and try to dump it to your board and this gonna work

i've done the same steps i've mentioned above and worked for me.
i hope this helps for someone.

BhavyeNijhawan, 2025-04-04:
> Hola a todos, tenia el mismo problema con un ESP32 CAM, buscando en otros foros ya nos habían otorgado la solución Solución:
>
> Hay que editar el archivo confi.ini de Thonny.
>
> Y en la parte de [ESP32] [...]

BRO YOU ARE A LIFE SAVER
EVEN THO I HAD TO GET IT TRANSLATED FROM GOOGLE
YOURS WAS THE ONLY METHOD THAT WORKED
THANK YOU SO MUCH!