thonny/thonny #2561: CircuitPython: Very long file saving time

LovelyA72, 2022-11-28:
It take a very long time for Thonny to save to a CircuitPython device when doing the first save after launch.
[image]

Thonny running on Windows 10 21H2 Python 3.10.6 64-bit

CircuitPython device: Raspberry Pi Pico W running 8.0.0-beta4

when saving, these texts will appear in the terminal running thonny:
```
14:19:40.735 INFO    thonny.plugins.micropython.mp_front: Falling back to serial.tools.list_ports.comports
14:19:43.247 INFO    thonny.ui_utils: Parent y: 50, rooty: 125, vrooty: 0
14:19:43.247 INFO    thonny.ui_utils: Placing .!inlinecommanddialog with geometry +813+566
14:19:44.662 INFO    thonny.ui_utils: Parent y: 50, rooty: 125, vrooty: 0
14:19:44.663 INFO    thonny.ui_utils: Placing .!inlinecommanddialog2 with geometry +813+566
```

aivarannamaa, 2022-12-05:
Thank you for the report!

Unfortunately I can't reproduce your issue. It could be helpful if you sent me the backend.log from Thonny's data directory:

* Select "Open Thonny data folder" from "Tools" menu.
* Reproduce the problem in your computer.
* After the problematic save has completed, close Thonny (this will finalize the log file)
* Post the backend.log from the data folder (feel free to review it first and remove any information you don't want to share, e.g. specific folder names). 
* (It wouldn't hurt if you also posted the frontend.log, but most likely backend.log will suffice)

LovelyA72, 2023-01-26:
Update: I might know the reason. I have a bunch of network drives(smb) mounted. I think Thonny is wasting it time on trying to connect drives that are not connected. I will investigate more.

LovelyA72, 2023-01-26:
Can confirm that with no network mapped drive delays are no longer exists.

aivarannamaa, 2023-01-26:
Thanks for this information! I'll try to optimize this.

Dan-Essin, 2024-04-30:
April 30, 2024
Still a problem