thonny/thonny #2564: Permissions error opening from networked drive

domluther, 2022-12-02:
Students at my school can have issues opening files in Thonny if they try to access their work via a Documents link in Quick Access link, it uses a networked path eg \\\\SHS-FILE01\\2020IntakeHome$\\... however if they go This PC and use the local path eg 'N:\Documents\Computing\Programming\lesson 1.py' then it works perfectly.

```
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 964, in dispatch_from_menu
    dispatch(None)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\workbench.py", line 908, in dispatch
    handler()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py", line 711, in _cmd_open_file
    self.show_file(path)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py", line 823, in show_file
    editor = self.get_editor(filename, True)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py", line 874, in get_editor
    return self._open_file(filename)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py", line 863, in _open_file
    editor._load_file(filename)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py", line 160, in _load_file
    self._load_local_file(filename, keep_undo)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py", line 182, in _load_local_file
    self._last_known_mtime = os.path.getmtime(self._filename)
  File "C:\Program Files (x86)\Thonny\lib\genericpath.py", line 55, in getmtime
    return os.stat(filename).st_mtime
PermissionError: [WinError 5] Access is denied: '\\\\SHS-FILE01\\2020IntakeHome$\\'

```

Is there something on the Thonny end that might be causing this? Is there an easy way to debug that?

aivarannamaa, 2022-12-05:
Thank you for the report!

Which Thonny version are you using? The stacktrace implies it's something way older than the current 4.0.1. In this case I recommend to try the newest version first.

domluther, 2022-12-07:
I'm on Thonny 4.0.1. If the stacktrace implies something else, then does it mean it might be somehow misconfigured?

aivarannamaa, 2022-12-28:
Sorry for the late reply!

Thonny 4 is not supposed to have the `C:\Program Files (x86)\Thonny\lib\site-packages\thonny\code.py` anymore. Could you re-check from "Help => About" that you really have version 4.0.1?

aivarannamaa, 2023-01-28:
Please re-open if you still have this problem and have more information to add.

closed 2023-01-28 by aivarannamaa

domluther, 2023-01-28:
Sorry for not replying sooner. Your answer was right. The issue was that the deployment hadn't worked properly. They had pushed out 4.0.1 but a few machines hadn't installed properly and so were on v3. Had completely missed that (should have seen due to the lack of the Ukrainian flag :))