thonny/thonny #2572: add option to remove support ukraine button or remove it

tigercoding56, 2022-12-15:
self explanatory  : i want to be in control of  the software i use on my system  
there is no excuse to try & force people to get into a conflict  also   
what does python have todo with conflicts : nothing   its just going 
to annoy people    
(also i am not pro-(anyone)  )

LexxM3, 2022-12-16:
While we should stay away from politics, keep in mind that you started it.  The simple summary is you have a choice not to use the free software provided to you if you don't agree with its politics or if annoys you in any other way -- I think that should be your approach here if you feel sufficiently strongly about the issue.  As you say, you want to have (and DO have) control over the software on your system.

For the rest of us ... this is obviously a critical issue to the primary Thonny author (and many others) and the attention he is drawing to it is quite restrained considering.  If he chooses to remove or give option to remove the support button per your request, it is up to him, but I would not support any of us attempting to dictate this to him.  Particularly not with the level of "empathy" shown in your request.

feecat, 2022-12-20:
Hi
I respect the author's decision, but I also think that there should be no politics in distributing software.  
Just a solution, After install, edit ..\Thonny\Lib\site-packages\thonny\workbench.py, locate line 690, remove following line.  

https://github.com/thonny/thonny/blob/6f41d84c07484af4bdc4a7e333e12149e3185cf9/thonny/workbench.py#L691-L700

TaChao, 2022-12-21:
> While we should stay away from politics, keep in mind that you started it. The simple summary is you have a choice not to use the free software provided to you if you don't agree with its politics [...]

Is this a threaten?

aivarannamaa, 2022-12-26:
I'll remove the button after Russian troops leave Ukraine. I won't make the button optional.

closed 2022-12-26 by aivarannamaa

makismantas, 2023-01-08:
feecat: Thanks! It worked!

maksim07, 2023-01-21:
@feecat thank much ! that worked

rfeng74, 2023-02-03:
@feecat  Thank you!

xkonglong, 2023-02-03:
I hope you can also support Syria refugees, Afghanistan refugees and all countries invaded by the United States. Thank you.

xkonglong, 2023-02-03:
> I'll remove the button after Russian troops leave Ukraine. I won't make the button optional.

 I hope you can also support Syria refugees, Afghanistan refugees and all countries invaded by the United States. Thank you.

khanhhua, 2023-02-21:
And the Iraqis as well. That Bush made joke about WMD which was never found in our face without a drop of shame.

Hottys, 2023-02-27:
Ukrainian Nazis thank the author. Donetsk apparently bombed itself in 2014 🤡🤡🤡🤡🤡

ipr87, 2023-03-04:
> Hi I respect the author's decision, but I also think that there should be no politics in distributing software. Just a solution, After install, edit ..\Thonny\Lib\site-packages\thonny\workbench.py, [...]

Thank you!

Rhadamanthys2, 2023-03-27:
> I'll remove the button after Russian troops leave Ukraine. I won't make the button optional.

This is clearly indoctrinating spam

jorgelex008, 2023-05-17:
@feecat Excellent. Worked perfectly. Thank you!

rn2208, 2023-06-01:
@feecat Thanks, you save my time.

st1lus1, 2023-06-13:
Hi! Do u know how I can do it on Mac OS?

stetsonacademy, 2023-09-20:
We're using this software to teach elementary school students the basics of python. They are too young to understand geopolitics, the history of Ukraine, or the atrocities of war. While many of would agree with the sentiment that any invasion is reprehensible, we resent being force-fed any political opinion

rn2208, 2023-09-20:
@stetsonacademy Hello, I have several students and now we use [pyzo](https://github.com/pyzo/pyzo/). This editor has everything you need for novice programmers.

rodolforizzo76, 2023-12-13:
> Hi I respect the author's decision, but I also think that there should be no politics in distributing software. Just a solution, After install, edit ..\Thonny\Lib\site-packages\thonny\workbench.py, [...]

Thanks for this solution, much appreciated. It was really something I didn't want to see. A questionable choice.

thatsunprofessional, 2024-06-05:
> Hi I respect the author's decision, but I also think that there should be no politics in distributing software. Just a solution, After install, edit ..\Thonny\Lib\site-packages\thonny\workbench.py, [...]

Thanks!!

JanisPlayer, 2024-08-27:
You are promoting a recommended program for the Raspberry Pico W, which is intended for elementary school children. So, am I understanding this correctly? You want to collect military donations here? Are you suggesting that the children should donate their toy weapons to soldiers, some of whom are not fighting voluntarily in Ukraine?

I’m from Germany, where we have the right to conscientious objection. In Ukraine, there’s a lot of corruption, and those who don’t want to fight have no choice or must pay. People are drafted at a young age, starting at 18, and come back with war trauma, PTSD. That’s way too young, but it’s well-known that young people are easier to convince to kill someone. On the battlefield, like in all wars, there are drugs and money to boost morale.

I’m also aware that conscription is sometimes used as an alternative to a fine, and that people are drafted indiscriminately. This, by the way, is also linked to the corruption in the country. So, only people who need money for their families fight, or are forced to with violence. Most of those with money have already left the country. In Germany, there have already been politicians who wanted to extradite conscientious objectors. Why? It's simple: we have interests. In other wars in Europe, we didn’t care; we didn’t even help democratic countries as the EU, because it didn’t concern us.

We’re not concerned with humanity; otherwise, we would support conscientious objection and not the idea of forcing young people, at an age where development and maturity are still in progress and where they’re easily influenced, into war. For us, this is more about geopolitics, weakening the Russian government and its sick war and imperial ambitions so that they become less of a threat to us—I get it.

I find it questionable to collect donations from elementary school children for such a cause. You wanted to mix politics or rather military donations into a program, and now you’re facing justified criticism.

So, do you want a Nerf gun so that fewer young people might die? But we always need to be as brutal as possible, don’t we?

My sincere suggestion: no donations for weapons or the military, but rather for humanitarian aid—the people are suffering enough. Whether victory is achieved or not, this war will bring significant losses to the population, and it already has.

Donations for weapons and the military are only okay if there’s also a right to conscientious objection. If you now tell me that Russia doesn’t have this either, let me tell you something. I don’t support the Russians; they have nothing to do with my European values, so this false argument with a poor comparison doesn’t work.

Therefore, the button is only okay for humanitarian aid, not for weapons in a war without the right to conscientious objection, because those holding the weapons don’t have a say.

Or another suggestion: a button against violence towards youth and for conscientious objection in Ukraine through a petition. Then you can keep the button, as it would be a statement that you do not support the forced conscription of young people, which can lead to psychological, physical harm, and even involuntary (coerced) death (murder) with these weapons.

I hope you like my opinion, because I had to listen to your opinion on coercion and murder too. Have you actually looked into what these weapons are used for and by whom? I think not, otherwise, you wouldn’t have included the button. Really, it’s questionable. I hope your statement supports conscientious objection and not “here are weapons, let the children with beards fight, and if they don’t do what they’re told, they’ll have no money and can expect violence.”

aivarannamaa, 2024-09-30:
> You are promoting a recommended program for the Raspberry Pico W, which is intended for elementary school children. 

Thonny has been designed mainly for university-level learners. I believe RPi chose Thonny over Mu for mainly technical reasons, as Thonny's default settings on RPi hide the parts which were the main reasons for creating it.

> Therefore, the button is only okay for humanitarian aid, not for weapons in a war without the right to conscientious objection, because those holding the weapons don’t have a say.

I re-checked -- https://github.com/thonny/thonny/wiki/Support-Ukraine clearly says "There are several options for supporting them", and donations to the military is just one of the proposed options. I believe most of the Thonny users visiting this page are mature enough to make their own choices and the number of children who make donations to the military because of this page, is rather small.

generalk25, 2024-10-07:
> Hi I respect the author's decision, but I also think that there should be no politics in distributing software. Just a solution, After install, edit ..\Thonny\Lib\site-packages\thonny\workbench.py, [...]

The real hero.

chriskriel, 2024-10-23:
The degree of entitlement to the effort of others displayed in this issue is astounding. If you are not comfortable with the author's motivation for doing this work, write your own of find another alternative.

makismantas, 2024-10-24:
@chriskriel I don't think you understand the concept of open software...

Vicrorege, 2025-01-06:
@feecat Thanks a lot! You save my time!

ToBowlRu, 2025-01-10:
@feecat Thank you!! I'm tired of people happily advocating for war and countless unnecessary deaths.

For those who are on Ubuntu 24.04 and install through apt, you'll find the workbench.py under
`/usr/lib/python3/dist-packages/thonny/workbench.py`
after that, its the same change.

aivarannamaa, 2025-01-10:
> I'm tired of people happily advocating for war and countless unnecessary deaths.

Would you bow down and seek a compromise if someone broke into your home and claimed it as theirs? Perhaps make an agreement that he can only rape your wife but leave your daughter alone? Or would you fight back? I'll stand with those who fight back. I advocate for stopping the aggressor.

This war was never necessary — not for the people whose lives have been taken or ruined, nor for the Russian people (except for the man with a dream of outdoing Peter the Great). But the act of fighting back is crucial. It's about avoiding countless of unnecessary deaths and ruined lives in the long run.

I'm not brave or skilled enough to fight in the battlefield, but I will at least express my support for those who are, instead of attacking the victim.

thatsunprofessional, 2025-01-10:
here's hoping for lasting peace which involves the Americans getting out of
Ukraine and as far away from Ukrainians as possible.

jorgelex008, 2025-01-11:
Way too much disinformation!

First: Keep politics aside.
Second: But if we must... Then read history! In depth, and beyond the
mainstream narrative. Real politics are ugly, and they are no innocents,
except those picked up from the streets and hauled into vans against their
will.
Third: People will see what they want to see, especially if emotionally (or
otherwise) invested. Keep an open mind and dig the facts carefully and
impartially.

Best regards to all.

aivarannamaa, 2025-01-11:
> here's hoping for lasting peace which involves the Americans getting out of Ukraine and as far away from Ukrainians as possible.

You are right, without American support, Ukraine would be peaceful as graveyard by now -- all opponents of the new power killed, in jail or intimidated to silence. For people in most countries, this kind of peace is not acceptable.

> Way too much disinformation!

Can you be more specific?

> Keep politics aside.

As I've said elsewhere, I refuse to call this politics. This may be politics for Russia, but for its neighbours it is a matter of life or death.

> Real politics are ugly,

What are the choices? Accepting the faith? Leaving large part of the population to soviet-style "peace"? Don't forget that without fight, this part would have been the whole Ukraine. War-time politics may be ugly, but justifying the agression in order to keep one's comfy lifestyle -- that's much uglier, that's really disgusting.

> People will see what they want to see, especially if emotionally (or otherwise) invested.

Apparently this also applies also to those, whose main worry is the rising price for natural gas.

thatsunprofessional, 2025-01-11:
like every word you have said is disinformation. ppl like you that push
this narrative are truly the worst enemy of Ukraine are the real ones
responsible for the death and loss and destruction and when it's over and
the Ukrainians that didn't sell out their own ppl will hate you more than
any Russian.

aivarannamaa, 2025-01-11:
I believe enough has been said here. I won't block the removal of the button for those who are happy to use an app dedicated to Ukraine but prefer to forget this. At the same time, I'll keep the button in the distribution until Russian troops leave Ukraine. The time will tell, whether its meaning will stay as call for help or it becomes a memory of an independent country, the reminder about the price of indifference.