thonny/thonny #2575: PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file'

j-herbig, 2022-12-20:
Hello,
I tried Thonny for the first time. I'm using an ESP32 and flashed MicroPython 1.19.1 on it. My code works when running it from Thonny Shell and also when using 'Run current script'. I tried saving the same code as 'main.py' on the ESP32, but this didn't work raising this error:

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (OSError: Could not open file /main.py for writing, output:
Traceback (most recent call last):

  File "<stdin>", line 4, in <module>

OSError: 36).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.
```

I also checked the backend.log:

```
22:41:27.546 INFO    thonny: Thonny version: 4.0.1
22:41:27.547 INFO    thonny: cwd: C:\Program Files (x86)\Thonny
22:41:27.547 INFO    thonny: original argv: ['C:\\Program Files (x86)\\Thonny\\python.exe', '-u', '-B', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM5', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:41:27.547 INFO    thonny: sys.executable: C:\Program Files (x86)\Thonny\python.exe
22:41:27.547 INFO    thonny: sys.argv: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp\\esp_back.py', "{'clean': True, 'port': 'COM5', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
22:41:27.547 INFO    thonny: sys.path: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\esp', 'C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages']
22:41:27.547 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
22:41:27.548 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': 'COM5', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'ESP32Proxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
22:41:27.559 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
22:41:27.559 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type EspMicroPythonBackend
22:41:27.562 INFO    thonny: TIME/MODS 0.320 before prepare (+73 modules)
22:41:27.562 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_ctypes', '_datetime', '_heapq', '_lzma', '_queue', '_struct', '_tkinter', 'binascii', 'bz2', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'msvcrt', 'pathlib', 'platform', 'posixpath', 'queue', 'serial', 'serial.serialutil', 'serial.serialwin32', 'serial.win32', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
22:41:27.562 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
22:41:27.562 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
22:41:27.632 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
22:41:27.645 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
22:41:27.652 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'OK\r\n'
22:41:27.654 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'MPY: soft reboot\r\n'
22:41:27.692 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
22:41:27.723 INFO    thonny: TIME/MODS 0.161 got welcome 
22:41:27.723 INFO    thonny: TIME/MODS 0.000 bef preparing helpers 
22:41:27.723 INFO    thonny.plugins.micropython.mp_back: Preparing helpers
22:41:27.724 INFO    thonny: TIME/MODS 0.001 befsubcode 
22:41:27.732 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
22:41:27.754 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
22:41:27.922 INFO    thonny: TIME/MODS 0.198 affsubcode 
22:41:27.957 INFO    thonny: TIME/MODS 0.036 affforw 
22:41:27.957 INFO    thonny: TIME/MODS 0.000 prepared helpers 
22:41:27.958 INFO    thonny: TIME/MODS 0.001 befsubcode 
22:41:27.972 INFO    thonny: TIME/MODS 0.013 affsubcode 
22:41:27.989 INFO    thonny: TIME/MODS 0.018 affforw 
22:41:27.989 INFO    thonny: TIME/MODS 0.000 got cwd 
22:41:27.990 INFO    thonny: TIME/MODS 0.001 befsubcode 
22:41:28.011 INFO    thonny: TIME/MODS 0.021 affsubcode 
22:41:28.036 INFO    thonny: TIME/MODS 0.025 affforw 
22:41:28.036 INFO    thonny: TIME/MODS 0.000 prepared 
22:41:28.036 INFO    thonny.plugins.micropython.mp_back: Prepared
22:41:28.036 INFO    thonny: TIME/MODS 0.000 befsubcode 
22:41:28.051 INFO    thonny: TIME/MODS 0.015 affsubcode 
22:41:28.161 INFO    thonny: TIME/MODS 0.110 affforw 
22:41:28.161 INFO    thonny: TIME/MODS 0.000 befsubcode 
22:41:28.210 INFO    thonny: TIME/MODS 0.049 affsubcode 
22:41:28.238 INFO    thonny: TIME/MODS 0.028 affforw 
22:41:28.238 INFO    thonny: TIME/MODS 0.000 befsubcode 
22:41:28.302 INFO    thonny: TIME/MODS 0.064 affsubcode 
22:41:28.332 INFO    thonny: TIME/MODS 0.030 affforw 
22:41:28.332 INFO    thonny.backend: Loading Jedi
22:41:28.333 INFO    thonny: TIME/MODS 0.000 Before loading Jedi 
22:41:28.333 INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages', 'C:\\Users\\johan\\AppData\\Roaming\\Thonny\\plugins\\Python310\\site-packages']
22:41:28.765 INFO    thonny: TIME/MODS 0.433 After loading Jedi (+123 modules)
22:41:28.766 INFO    thonny: NEW MODS ['_blake2', '_compat_pickle', '_hashlib', '_json', '_pickle', '_socket', 'difflib', 'filecmp', 'gc', 'hashlib', 'importlib._abc', 'importlib.util', 'jedi', 'jedi._compatibility', 'jedi.api', 'jedi.api.classes', 'jedi.api.completion', 'jedi.api.completion_cache', 'jedi.api.environment', 'jedi.api.errors', 'jedi.api.exceptions', 'jedi.api.file_name', 'jedi.api.helpers', 'jedi.api.interpreter', 'jedi.api.keywords', 'jedi.api.project', 'jedi.api.refactoring', 'jedi.api.refactoring.extract', 'jedi.api.strings', 'jedi.cache', 'jedi.common', 'jedi.debug', 'jedi.file_io', 'jedi.inference', 'jedi.inference.analysis', 'jedi.inference.arguments', 'jedi.inference.base_value', 'jedi.inference.cache', 'jedi.inference.compiled', 'jedi.inference.compiled.access', 'jedi.inference.compiled.getattr_static', 'jedi.inference.compiled.mixed', 'jedi.inference.compiled.subprocess', 'jedi.inference.compiled.subprocess.functions', 'jedi.inference.compiled.value', 'jedi.inference.context', 'jedi.inference.docstring_utils', 'jedi.inference.docstrings', 'jedi.inference.filters', 'jedi.inference.flow_analysis', 'jedi.inference.gradual', 'jedi.inference.gradual.annotation', 'jedi.inference.gradual.base', 'jedi. [... cut ...]
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1267, in _write_file
    self._write_file_via_serial(source_fp, target_path, file_size, callback)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 1327, in _write_file_via_serial
    raise OSError(
OSError: Could not open file /main.py for writing, output:
Traceback (most recent call last):

  File "<stdin>", line 4, in <module>

OSError: 36
```

aivarannamaa, 2022-12-28:
Thank you for the report!

I haven't seen Error 36 myself, but according to internet search it should be related to too long path name, which is surprising considering the path `/main.py`.

Can you please tell me the output of following commands:

```python
import os
import sys
print(os.listdir("/"))
print(sys.path)
```

Can you tell me the exact MicroPython variant and your board model?

j-herbig, 2023-01-06:
Hi, thank you for your reply. I tried another ESP32, exactly same model, and there it worked as it should. I still don't know, what exactly caused the error. It looks like the ESP32 was responsible and Thonny is working fine.

aivarannamaa, 2023-01-06:
Note that there are several variants of MicroPython for ESP32, one difference being the expected flash size. You may want to investigate whether you had correct variant on your first board.

closed 2023-01-06 by aivarannamaa