thonny/thonny #2576: Unable to flash micropython in Pi Pico

Anshul-13J, 2022-12-21:
Getting the error:
`Could not download variants info from https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json
`
[image]

Contents of my INFO_UF2.txt file:
`UF2 Bootloader v2.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2
`

I'm using Thonny version 4.0.1

katerasrael, 2022-12-22:
Same here

Windows 10
Thonny 4.0.1

aivarannamaa, 2022-12-26:
Thank you for the report!

Could you please send me the error part (or whole content) from frontend.log (Tools => Open Thonny data folder)?

Vasudevatirupathinaidu, 2022-12-27:
I'm facing the same issue.

Windows 8.1
Thonny 4.0.1

When I try to download the UF2 file from the official page, a file with the name "rp2-pico-20221220-unstable-v1.19.1-782-g699477d12.uf2" has been downloaded. Does this unstable file cause this issue?
[image]

Anshul-13J, 2022-12-27:
> Thank you for the report!
> 
> Could you please send me the error part (or whole content) from frontend.log (Tools => Open Thonny data folder)?

Yes, I'm attaching the contents of frontend.log file.

`11:55:51.207 INFO    thonny: Thonny version: 4.0.1
11:55:51.207 INFO    thonny: cwd: C:\Program Files (x86)\Thonny
11:55:51.207 INFO    thonny: original argv: []
11:55:51.207 INFO    thonny: sys.executable: C:\Program Files (x86)\Thonny\pythonw.exe
11:55:51.207 INFO    thonny: sys.argv: ['C:\\Program Files (x86)\\Thonny\\thonny.exe']
11:55:51.207 INFO    thonny: sys.path: ['C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages']
11:55:51.207 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
11:55:57.928 INFO    thonny.shell: BaseShellText._on_backend_terminated
`

Or, you can find the file attached:
[frontend.log](https://github.com/thonny/thonny/files/10306324/frontend.log)

aivarannamaa, 2022-12-27:
@Anshul-13J, thanks, but it looks like you took this log file right after starting Thonny. Please first try to reproduce the error you're receiving and then take the log file. Thonny creates a fresh log file at each launch.

Anshul-13J, 2022-12-27:
[frontend.log](https://github.com/thonny/thonny/files/10309397/frontend.log)

Here's the updated file.
If you're looking for the logs:
`21:12:00.862 INFO    thonny: Thonny version: 4.0.1
21:12:00.862 INFO    thonny: cwd: C:\Program Files (x86)\Thonny
21:12:00.862 INFO    thonny: original argv: []
21:12:00.862 INFO    thonny: sys.executable: C:\Program Files (x86)\Thonny\pythonw.exe
21:12:00.862 INFO    thonny: sys.argv: ['C:\\Program Files (x86)\\Thonny\\thonny.exe']
21:12:00.862 INFO    thonny: sys.path: ['C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages']
21:12:00.862 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=1, no_site=0, ignore_environment=1, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=1, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
21:12:06.313 INFO    thonny.shell: BaseShellText._on_backend_terminated
21:12:14.643 INFO    thonny.plugins.micropython.uf2dialog: Downloading 'https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json'
21:12:14.695 INFO    thonny.ui_utils: Parent y: 155, rooty: 218, vrooty: 0
21:12:14.695 INFO    thonny.ui_utils: Placing .!uf2flashingdialog with geometry +367+233
21:12:56.919 ERROR   thonny.plugins.micropython.uf2dialog: Could not download variants info from https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 1348, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 1282, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 1328, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 1277, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 1037, in _send_output
    self.send(msg)
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 975, in send
    self.connect()
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 1447, in connect
    super().connect()
  File "C:\Program Files (x86)\Thonny\lib\http\client.py", line 941, in connect
    self.sock = self._create_connection(
  File "C:\Program Files (x86)\Thonny\lib\socket.py", line 845, in create_connection
    raise err
  File "C:\Program Files (x86)\Thonny\lib\socket.py", line 833, in create_connection
    sock.connect(sa)
TimeoutError: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\uf2dialog.py", line 336, in _download_variants
    with urlopen(req) as fp:
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 519, in open
    response = self._open(req, data)
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 536, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 496, in _call_chain
    result = func(*args)
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 1391, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "C:\Program Files (x86)\Thonny\lib\urllib\request.py", line 1351, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond>
`

aivarannamaa, 2022-12-29:
@Anshul-13J, thanks for the log! 

Can you open https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json in your web browser? How long does it take to open it fully?

Anshul-13J, 2022-12-29:
> @Anshul-13J, thanks for the log!
> 
> Can you open https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json in your web browser? How long does it take to open it fully?

It doesn't load.

[image]

aivarannamaa, 2022-12-29:
Can you open https://github.com/thonny/thonny/blob/master/data/micropython-variants-uf2.json ? If yes, then can you open the "Raw" link on this page? If yes, then what is the URL of the "Raw" link?

Anshul-13J, 2022-12-29:
> Can you open https://github.com/thonny/thonny/blob/master/data/micropython-variants-uf2.json ? If yes, then can you open the "Raw" link on this page? If yes, then what is the URL of the "Raw" link?

Yes, I'm able to open  https://github.com/thonny/thonny/blob/master/data/micropython-variants-uf2.json .
But "raw" link is not opening, it's the same link: https://raw.githubusercontent.com/thonny/thonny/master/data/micropython-variants-uf2.json

I'm going to try it with VPN now, as I'm able to access the raw link using Tor

Anshul-13J, 2022-12-29:
UPDATE:

The issue is solved by using the VPN.
The raw github link was causing the problem but it worked when I used VPN.

closed 2022-12-29 by Anshul-13J

Ebershon, 2024-04-28:
[image]

Ebershon, 2024-04-28:
Me salio ese error estoy usando esp32. Probé con otro cable otro micro controlador pero sin éxito alguna solución? Tengo el Windows 11

joomab, 2024-05-14:
@Ebershon tengo el mismo error, lograste solucionar algo?