thonny/thonny #2578: Thonny 4.0.1 hangs (OSX) when expanding a output dict on Console

juanfal, 2022-12-21:
It needs to be killed by the System (as Thonny program, not as Python process)

aivarannamaa, 2022-12-22:
Thank you for the report!

If it is a very long line, then it's not easy to fix -- Tkinter's Text widget is notoriously slow with long lines. The best I can think of is saving the line into a file and have Thonny open it with an external text editor.

juanfal, 2022-12-22:
It's true! On a M1, Thonny seems not to crash, but getting really slow showing the output… it really seems a bug in Tkinter. Even on M1 the spinning colourful ball appears when you try to interact with the output
Sending the program:

[comprimido.zip](https://github.com/thonny/thonny/files/10285542/comprimido.zip)

(in exams, it there are always problems with students getting crashes when executing not saved files and others not finding the data file to open when they execute the on-memory-buffer .py program)