thonny/thonny #2584: More Detailed Thonny Plugin Guide / Wiki

limitcracker, 2022-12-25:
I would like to make a Thonny Plugin in order to make two PanedWindows for fetching assignments form local/remote APIs.

[image]

Is there a more detailed guide on how to

- understand more easily the layout geometry
- make submenus
- make PanedWindows
- showing reports (i.e. pytest etc)
?

I managed to make a fetch working example by studying a 3rd party plugin but I find it a little bit difficult to proceed on the aforementioned topics.

Any ideas?

limitcracker, 2022-12-25:
My working example that fetches a **.py** file assignment an loads it in editor:
```

#import os
#import sys
import urllib.request
from collections import namedtuple
from tkinter.messagebox import showinfo

from thonny import get_workbench

name = "thonny-exercise-loader"

ErrorMessage = namedtuple("ErrorMessage", ["error_type", "description"])  # NOSONAR

SUCCESS = "All done!"

NO_TEXT_TO_FORMAT = ErrorMessage("Nothing to do here", "There is no text to format.")
PACKAGE_NOT_FOUND = ErrorMessage(
    "Package not found",
    "Could not find Black package. Is it installed and on your PATH?",
)
NOT_COMPATIBLE = ErrorMessage(
    "File not compatible!",
    "Looks like this is not a python file. Did you already save it?",
)

class Exercise:
    """
    Fetch exercise to the current loaded file.

    Description
    """

    def __init__(self) -> None:
        """Get the workbench to fetch and copy the fetched content to editor, later."""
        self.workbench = get_workbench()

    def load_exercise(self) -> None:
        """Handle the plugin execution."""
        self.editor = self.workbench.get_editor_notebook().get_current_editor()

        self.filename = self.editor.get_filename()
        if self.filename is not None and self.filename[-3:] == ".py":
            self.editor.save_file()

            self.filename = "/Users/limitcracker/Documents/Projects/ThonnyPlugin/test.py"
            # self.filename = "https://raw.githubusercontent.com/thonny/thonny/3eaa0319a9722bcdac12b4401b0a69b47d2e9f00/thonny/first_run.py"

            file_url = 'https://raw.githubusercontent.com/thonny/thonny/3eaa0319a9722bcdac12b4401b0a69b47d2e9f00/thonny/first_run.py'
            content = urllib.request.urlopen(file_url).read().decode('utf-8')
            self.file_handle = open(self.filename, "w")
            self.file_handle.write(content)
            self.file_handle.close()
            

            # self.editor._load_file(self.filename, keep_undo=True)
            # self.editor._load_remote_file(self.filename)

        showinfo(title="Exercise Loader", message="OK")

    def load_plugin(self) -> None:
        """
        Load the plugin on runtime.

        Using self.workbench.add_command(), the plugin is registered in Thonny
        with all the given arguments.
        """

        self.workbench.add_command(
            command_id="load_exercise",
            menu_name="tools",
            command_label="Load Exercise",
            handler=self.load_exercise,
            default_sequence="<Control-Alt-c>",
            extra_sequences=["<<CtrlAltCInText>>"],
            submenu=self.tools_load_exercise,
        )

if get_workbench() is not None:
    run = Exercise().load_plugin()

```

aivarannamaa, 2022-12-26:
Hi!

I suppose you have found https://github.com/thonny/thonny/wiki/Plugins. Unfortunately I don't have anything more detailed, beside examples.

Fortunately, there is an existing plug-in for an exercise system, which you may be able to customize for your needs: https://github.com/kspar/thonny-easy

It is divided into the UI part and provider-part. The UI part is meant to be generic enough to be suitable for many exercise systems. You may be able to make it work for you by implementing custom provider part. Here's a simple example: https://github.com/kspar/thonny-easy/blob/master/thonnycontrib/easy/demo_exercise_provider.py and here is a real provider: https://github.com/kspar/thonny-easy/blob/master/thonnycontrib/easy/easy_provider.py

closed 2022-12-27 by limitcracker