thonny/thonny #2585: The device does not recognize the command. Error

Komas19-new, 2022-12-27:
```
PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
```

I have seen other people have the issue, and resetting flash memory made it work again, but the issue is, i don't have any backup of the files inside the flash memory. Is there any way to get it?

aivarannamaa, 2022-12-27:
When do you get this error? On saving a file? It could help if you posted your backend.log (Tools => Open Thonny data folder) right after getting this error.

Komas19-new, 2022-12-27:
I have just randomly fixed it, i spammed the stop button and it worked again, it's weird that this happended, the error occured again, but pressing STOP twice worked, maybe it's the usb cable doing something? because the raspberry pi pico w keeps on getting disconnected.

Here is the backend.log, because it's better to fix it for good.
[backend.log](https://github.com/thonny/thonny/files/10309377/backend.log)

Komas19-new, 2022-12-27:
i am gonna take this chance to copy all data

aivarannamaa, 2023-01-28:
Sorry for the late reply!

Unfortunately the log doesn't show this error. Thonny overwrites the backend.log each time you reconnect or Stop/Restart the interpreter.