thonny/thonny #2587: error when trying to import matplotlib library

thiago-user, 2022-12-29:
I installed the matplotlib library in Thonny, but when I try to import it, I get this error: from matplotlib._path import (
ImportError: DLL load failed while importing _path: Could not find the specified module.

aivarannamaa, 2022-12-29:
Thank you for the report!

Which OS are you using? 32-bit or 64-bit? Which Thonny version? Did you install it via "Tools => Manage packages" ?

thiago-user, 2022-12-29:
> Thank you for the report!
> 
> Which OS are you using? 32-bit or 64-bit? Which Thonny version? Did you install it via "Tools => Manage packages" ?

Hi. My Windows is 64 bits. thonny is 4.0.1 and i installed it using package manager

aivarannamaa, 2022-12-29:
Could you please open the package manager again, select `matplotlib` from the list in the left and post the screenshot of the dialog?

thiago-user, 2022-12-29:
> Could you please open the package manager again, select `matplotlib` from the list in the left and post the screenshot of the dialog?

[image]

aivarannamaa, 2022-12-29:
Please check whether you have another copy of `matplotlib` in `C:\Users\Thiago\AppData\Local\Programs\Thonny\Lib\site-packages`

thiago-user, 2022-12-29:
> Please check whether you have another copy of `matplotlib` in `C:\Users\Thiago\AppData\Local\Programs\Thonny\Lib\site-packages`

would it be this?

[image]

aivarannamaa, 2022-12-29:
I don't know if this is relevant but I'm surprised that you have such a lot of packages under Thonny's installation directory. The packages installed via "Tools => Manage packages" are not supposed to end up in this directory.

If you experimented with different means of installing packages (eg. via Thonny's command line) there may be some kind of conflicts between different packages. 

I would proceed by deleting `C:\Users\Thiago\AppData\Local\Programs\Thonny` and re-installing Thonny.

aivarannamaa, 2022-12-29:
Oh, one more thing... Did you install XXL variant of Thonny or the regular one?

thiago-user, 2022-12-29:
Dude, I just formatted the computer. I installed matplotlib and, as it gave an error, I also installed two more packages just to test if they also had the same error. so I think that all these installed packages are standard for thonny (or python, which I also just installed).

As for thonny's version, I have no idea what this xxl is. I downloaded the version from the first link that appears on the official page thonny.org

thiago-user, 2022-12-29:
once I installed thonny, I also installed matplotlib via package manager. I had absolutely nothing related to programming installed on my computer until then. I believe it is not conflict with other packages, I think. the only thing I found on google is that the error may be related to environment variables, but I didn't quite understand

thiago-user, 2022-12-29:
> I don't know if this is relevant but I'm surprised that you have such a lot of packages under Thonny's installation directory. The packages installed via "Tools => Manage packages" are not supposed [...]

the suggestions didn't work :(

aivarannamaa, 2022-12-29:
Please run this diagnostic script and post the output:

```python
import sys
import os.path

print("sys.path:")
for p in sys.path:
    if p == "":
        p = os.getcwd()
    print(f"folder {p}:")
    if os.path.isdir(p):
        for name in os.listdir(p):
            if name.startswith("matplotlib"):
                print(f"  contains {name}")

import matplotlib
print(matplotlib.__file__)
from matplotlib import pyplot
print(pyplot.__file__)
```

thiago-user, 2022-12-29:
> Please run this diagnostic script and post the output:
>
> ```python
> import sys
> import os.path
>
> print("sys.path:")
> for p in sys.path:
>     if p == "":
>         p = os.getcwd()
>     print(f"folder [...]

sys.path:
folder C:\Users\Thiago\Desktop:
folder C:\Users\Thiago\AppData\Local\Programs\Thonny\python310.zip:
folder C:\Users\Thiago\AppData\Local\Programs\Thonny\DLLs:
folder C:\Users\Thiago\AppData\Local\Programs\Thonny\lib:
folder C:\Users\Thiago\AppData\Local\Programs\Thonny:
folder C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages:
  contains matplotlib
  contains matplotlib-3.6.2-py3.10-nspkg.pth
  contains matplotlib-3.6.2.dist-info
folder C:\Users\Thiago\AppData\Local\Programs\Thonny\lib\site-packages:
Traceback (most recent call last):
  File "C:\Users\Thiago\Desktop\diagnostic.py", line 14, in <module>
    import matplotlib
  File "C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages\matplotlib\__init__.py", line 113, in <module>
    from . import _api, _version, cbook, _docstring, rcsetup
  File "C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages\matplotlib\rcsetup.py", line 27, in <module>
    from matplotlib.colors import Colormap, is_color_like
  File "C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages\matplotlib\colors.py", line 56, in <module>
    from matplotlib import _api, _cm, cbook, scale
  File "C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages\matplotlib\scale.py", line 22, in <module>
    from matplotlib.ticker import (
  File "C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages\matplotlib\ticker.py", line 138, in <module>
    from matplotlib import transforms as mtransforms
  File "C:\Users\Thiago\AppData\Roaming\Python\Python310\site-packages\matplotlib\transforms.py", line 49, in <module>
    from matplotlib._path import (
ImportError: DLL load failed while importing _path: Não foi possível encontrar o módulo especificado.
>>>

aivarannamaa, 2022-12-30:
Try installing Visual C++ redistributable: https://download.visualstudio.microsoft.com/download/pr/bcb0cef1-f8cb-4311-8a5c-650a5b694eab/2257B3FBE3C7559DE8B31170155A433FAF5B83829E67C589D5674FF086B868B9/VC_redist.x64.exe

thiago-user, 2022-12-30:
It worked! thank you friend. thank you very much for your time and your help. I had no idea how to solve this

thiago-user, 2022-12-30:
but I already had windows 10 installed and this problem had not happened. why did it happen now?

aivarannamaa, 2022-12-30:
> I already had windows 10 installed and this problem had not happened. why did it happen now?

Visual C++ redistributable is not installed by default. It gets installed together with some programs automatically, that's why people usually don't remember installing it.

closed 2022-12-30 by aivarannamaa