thonny/thonny #2588: Raspberry Pi Pico - First project problem

ThiNguyen3008, 2022-12-29:
Hello, I'm a beginner for Raspberry Pi Pico. I cannot run my first project and I don't know why. I'm using Thonny 4.0.1. Can you help me please? Thank you
This is from the book
[image]
This is a screenshot of my code
[image]

samArtDev, 2022-12-30:
Did you set your interpreter in Thonny's options? "Local Python 3" is different than MicroPython.

[image]

aivarannamaa, 2022-12-30:
@ThiNguyen3008, you haven't paid attention to the indentation of the lines below `while`.

samArtDev, 2022-12-30:
I completely didn't notice their mistake. Sorry. Just for clarification to new programmers, Python doesn't require semicolons ( ; ) the same way other programming languages do. The indents control the context for the code.

[image]

closed 2023-01-28 by aivarannamaa