thonny/thonny #2589: Thonny can't read Raspberry Pi Pico W

Nicrom098195, 2022-12-30:
Hello all!

I was trying to make a web server with my raspberry pi pico W, and it was working very well, but when i was making a new version of the code, Thonny started to doesn't read Raspberry Pi.
I tried to reset it, and when i reset it Thonny has installed MicroPython on it, but after the installation the error was already here.
I tried already on a Raspberry Pi Pico H, and it was working good.
The error is:
```
PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialException: write failed: [Errno 5] Input/output error).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0.
```
I'm using Linux Mint 20.3 Cinnamon.

aivarannamaa, 2022-12-30:
Thank you for the report!

Please select "Tools => Open Thonny data folder", reproduce this error and post the contents of backend.log.

Nicrom098195, 2022-12-30:
That's the backend.log's content:
```
20:02:06.895 INFO    thonny: Thonny version: 4.0.1
20:02:06.895 INFO    thonny: cwd: /home/nicrom09
20:02:06.896 INFO    thonny: original argv: ['/home/nicrom09/apps/thonny/bin/python3.10', '-u', '-B', '/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': False, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RP2040BackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
20:02:06.896 INFO    thonny: sys.executable: /home/nicrom09/apps/thonny/bin/python3.10
20:02:06.896 INFO    thonny: sys.argv: ['/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': False, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RP2040BackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
20:02:06.896 INFO    thonny: sys.path: ['/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/rp2040', '/home/nicrom09/apps/thonny/lib/python310.zip', '/home/nicrom09/apps/thonny/lib/python3.10', '/home/nicrom09/apps/thonny/lib/python3.10/lib-dynload', '/home/nicrom09/apps/thonny/lib/python3.10/site-packages']
20:02:06.896 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
20:02:06.896 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': '/dev/ttyACM0', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RP2040BackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
20:02:06.898 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 64, write_block_delay: 0, read_block_size: 0
20:02:06.898 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type RP2040Backend
20:02:06.899 INFO    thonny: TIME/MODS 0.067 before prepare (+73 modules)
20:02:06.899 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
20:02:06.900 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
20:02:06.900 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
20:02:06.950 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
20:02:07.002 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
20:02:07.006 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
20:02:07.108 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
20:02:07.108 ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 139, in __init__
    self._process_until_initial_prompt(
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 268, in _process_until_initial_prompt
    self._process_output_until_active_prompt(
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/bare_metal_backend.py", line 844, in _process_output_until_active_prompt
    self._interrupt()
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 337, in _interrupt
    self._write(INTERRUPT_CMD)
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/mp_back.py", line 466, in _write
    return self._connection.write(data)
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/thonny/plugins/micropython/serial_connection.py", line 89, in write
    size = self._serial.write(data)
  File "/home/nicrom09/apps/thonny/lib/python3.10/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 5] Input/output error

```

aivarannamaa, 2023-01-01:
The error happens during Thonny connecting to your board, right?

Which MicroPython version are you using?

Nicrom098195, 2023-01-02:
MicroPython *20221220-unstable-1.19.1*

aivarannamaa, 2023-01-28:
Sorry for the late reply!

In case you still need solution to this problem, I suggest trying a newer daily build of MicroPython.

Nicrom098195, 2023-01-29:
I tried to change version, but it give me the same error.
Is possible that the problem is my Pico W?

aivarannamaa, 2023-02-12:
> Is possible that the problem is my Pico W?

Yes, this is a possibility.

saltare, 2023-02-12:
I had the same error. It might be something to do with needing to wipe main.py off it, see eg [this](https://forums.raspberrypi.com/viewtopic.php?t=305432.)
Try this https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory

Nicrom098195, 2023-02-18:
> I had the same error. It might be something to do with needing to wipe main.py off it, see eg [this](https://forums.raspberrypi.com/viewtopic.php?t=305432.) Try this https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory

This is also not working

Nicrom098195, 2023-02-18:
> I had the same error. It might be something to do with needing to wipe main.py off it, see eg [this](https://forums.raspberrypi.com/viewtopic.php?t=305432.) Try this https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#resetting-flash-memory

that didn't work either, but I tried using circuitpython and not micropython, and now it works perfectly!

Thank you very much for the help!

closed 2023-02-18 by Nicrom098195