thonny/thonny #2592: I2C problem with Rasberry Pi Pico W

PSonnie, 2022-12-31:
I want to use an OLED SSD1306 with the Raspberry Pi Pico W. However ik get a system error after I run a small program. I have test it on 2 Pico W en one Raspberry Pico (without W). The program runs okay on the Raspberry Pi Pico however not on te Pico W. I use a Windows 10 laptop with Thonny 4.01 communicating with the Pico W board was trough Com11 and Com13. Communication with the Pico board was trough Com8. I have also tried several SSD1306 drivers all downloaded from several sites on the internet with no succes. The error also shows up also when I use the same program with Thonny on a Raspberry Pi4 and Pi400 with a recent downloaded OS.
See the error on the link below:
[image]

PSonnie, 2022-12-31:
Additional info for I2C problem on the Raspberry Pi Pico W:
The pico W was running:  [v1.19.1-779-g57bb1e047 (2022-12-19) .uf2](https://micropython.org/resources/firmware/rp2-pico-w-20221219-unstable-v1.19.1-779-g57bb1e047.uf2)
and I tried also:
[v1.19.1-782-g699477d12 (2022-12-20) .uf2](https://micropython.org/resources/firmware/rp2-pico-w-20221220-unstable-v1.19.1-782-g699477d12.uf2)

aivarannamaa, 2023-01-01:
This doesn't look like Thonny-specific problem. Try https://github.com/orgs/micropython/discussions or https://forums.raspberrypi.com/viewforum.php?f=146

closed 2023-01-01 by aivarannamaa

PSonnie, 2023-01-01:
Okay thanks for the quick answer. Problem is already solved. Pins SCL and SDA were swapped in hardware. 
This means that the driver ssd1306 can check in one or the other way the hardware load connection of both output lines SCL and SDA of the raspberry Pi Pico W.