thonny/thonny #2593: Problem with package installation

Patrykchylinski, 2022-12-31:
Hi, I am having a huge problem installing any package. 
Here is an example of an error when installing a package. No matter which one I choose, it's always the same.

**"Error Command ['/Users/mac/Library/Caches/pipkin/workspaces/d64e986491885065d1f585e6f96486a/bin/python3','-I' -m', 'pip', '-no-color', '-disable-pip-version-check', '-trusted-host', '127.0.0.1', 'install', -no-compile','--use-pep517', --upgrade-strategy' 'only-if-needed','Adafruit-Blinka',"**

MacOS Monterey Version 12.6.2, Thonny 4.0.1, Darwin 21.6.0 (64-bit), Python 3.10.6 (64 bit), Tk 8.6.12

For the past two days I've been fiddling and figuring out what I can do.

aivarannamaa, 2023-01-01:
Thank you for the report!

It looks like you posted only the start of the error message. Please post the whole message.

Patrykchylinski, 2023-01-01:
> Thank you for the report!
> 
> It looks like you posted only the start of the error message. Please post the whole message.

I do not hide that I do not know how to copy the rest because copying does not work. Can I find the logs somewhere?

Patrykchylinski, 2023-01-03:
> 

Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [03/Jan/2023 15:38:27] "GET /adafruit-micropython-blinka/ HTTP/1.1" 200 -
Collecting Adafruit-Micropython-Blinka
127.0.0.1 - - [03/Jan/2023 15:38:27] "GET /adafruit-micropython-blinka/Adafruit-Micropython-Blinka-1.0.0.tar.gz/ HTTP/1.1" 200 -
  Downloading http://127.0.0.1:36628/adafruit-micropython-blinka/Adafruit-Micropython-Blinka-1.0.0.tar.gz/
     - 11.9 kB ? 0:00:00
  Installing build dependencies: started
127.0.0.1 - - [03/Jan/2023 15:38:30] "GET /setuptools/ HTTP/1.1" 200 -
127.0.0.1 - - [03/Jan/2023 15:38:30] "GET /setuptools/setuptools-65.6.3-py3-none-any.whl/ HTTP/1.1" 200 -
127.0.0.1 - - [03/Jan/2023 15:38:31] "GET /wheel/ HTTP/1.1" 200 -
127.0.0.1 - - [03/Jan/2023 15:38:31] "GET /wheel/wheel-0.38.4-py3-none-any.whl/ HTTP/1.1" 200 -
127.0.0.1 - - [03/Jan/2023 15:38:32] "GET /pip/ HTTP/1.1" 200 -
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'
  error: subprocess-exited-with-error
  
  Getting requirements to build wheel did not run successfully.
  exit code: 1
  
  [17 lines of output]
  Traceback (most recent call last):
    File "C:\Users\patryk.chylinski\AppData\Local\pipkin\cache\workspaces\696aa67e4ad739ba77b5803f608b243c\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 363, in <module>
      main()
    File "C:\Users\patryk.chylinski\AppData\Local\pipkin\cache\workspaces\696aa67e4ad739ba77b5803f608b243c\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "C:\Users\patryk.chylinski\AppData\Local\pipkin\cache\workspaces\696aa67e4ad739ba77b5803f608b243c\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 130, in get_requires_for_build_wheel
      return hook(config_settings)
    File "C:\Users\patryk.chylinski\AppData\Local\Temp\pip-build-env-d4beveiv\overlay\Lib\site-packages\setuptools\build_meta.py", line 338, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "C:\Users\patryk.chylinski\AppData\Local\Temp\pip-build-env-d4beveiv\overlay\Lib\site-packages\setuptools\build_meta.py", line 320, in _get_build_requires
      self.run_setup()
    File "C:\Users\patryk.chylinski\AppData\Local\Temp\pip-build-env-d4beveiv\overlay\Lib\site-packages\setuptools\build_meta.py", line 484, in run_setup
      super(_BuildMetaLegacyBackend,
    File "C:\Users\patryk.chylinski\AppData\Local\Temp\pip-build-env-d4beveiv\overlay\Lib\site-packages\setuptools\build_meta.py", line 335, in run_setup
      exec(code, locals())
    File "<string>", line 39, in <module>
  FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\patryk.chylinski\\AppData\\Local\\Temp\\pip-install-lsjcygbr\\adafruit-micropython-blinka_4d7b287f1bb6491db820babb98edeb9a\\src\\__version__.py'
  [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

Getting requirements to build wheel did not run successfully.
exit code: 1

See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
Error Command '['C:\\Users\\patryk.chylinski\\AppData\\Local\\pipkin\\cache\\workspaces\\696aa67e4ad739ba77b5803f608b243c\\Scripts\\python.exe', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'Adafruit-Micropython-Blinka',

odysseus55, 2023-01-04:
I am getting the same issue on linux mint and windows11 versions.

aivarannamaa, 2023-01-05:
@Patrykchylinski, do I understand properly that you tried to install `Adafruit-Micropython-Blinka` for your MicroPython board?

aivarannamaa, 2023-01-05:
Which MicroPython board are you using? Which MicroPython version?

ncmemory, 2023-01-19:
Hello I have the same problem under Windows 11 with the Thonny version 4.0.1. I'm trying to install micropython-umqtt.simple with PyPi and I get this error as well. My MicroPython board is a D1 mini ESP 8266  and my MicroPython version is esp8266-20220618-v1.19.1.bin. What can you do there?

Looking in indexes: http://127.0.0.1:36628
127.0.0.1 - - [19/Jan/2023 17:18:22] "GET /micropython-umqtt-simple/ HTTP/1.1" 200 -
Collecting micropython-umqtt.simple
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 56296)
Traceback (most recent call last):
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 204, in __init__
    super().__init__(request, client_address, server)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 747, in __init__
    self.handle()
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\http\server.py", line 432, in handle
    self.handle_one_request()
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\http\server.py", line 420, in handle_one_request
    method()
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 211, in do_GET
    self._serve_file(*path.split("/"))
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 245, in _serve_file
    tweaked_bytes = self._tweak_file(dist_name, file_name, original_bytes)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 312, in _tweak_file
    assert normalize_dist_name(wrapper_dir).startswith(normalize_dist_name(dist_name))
AssertionError
----------------------------------------
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /micropython-umqtt-simple/micropython-umqtt.simple-1.3.4.tar.gz/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 56298)
Traceback (most recent call last):
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 204, in __init__
    super().__init__(request, client_address, server)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 747, in __init__
    self.handle()
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\http\server.py", line 432, in handle
    self.handle_one_request()
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\http\server.py", line 420, in handle_one_request
    method()
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 211, in do_GET
    self._serve_file(*path.split("/"))
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 245, in _serve_file
    tweaked_bytes = self._tweak_file(dist_name, file_name, original_bytes)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 312, in _tweak_file
    assert normalize_dist_name(wrapper_dir).startswith(normalize_dist_name(dist_name))
AssertionError
----------------------------------------
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', RemoteDisconnected('Remote end closed connection without response'))': /micropython-umqtt-simple/micropython-umqtt.simple-1.3.4.tar.gz/
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 56300)
Traceback (most recent call last):
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Users\Memory\AppData\Local\Programs\Thonny\lib\site-packages\thonny\vendored_libs\pipkin\proxy.py", line 204, in __init__
[... cut ...] on aborted.', RemoteDisconnected('Remote end closed connection without response')))

Error Command '['C:\\Users\\Memory\\AppData\\Local\\pipkin\\cache\\workspaces\\9cb8c7a1e60e4427b74fd273d8cf33bc\\Scripts\\python.exe', '-I', '-m', 'pip', '--no-color', '--disable-pip-version-check', '--trusted-host', '127.0.0.1', 'install', '--no-compile', '--use-pep517', '--upgrade-strategy', 'only-if-needed', 'micropython-umqtt.simple', '--index-url', 'http://127.0.0.1:36628']' returned non-zero exit status 1.

ncmemory, 2023-01-19:
Thanks for the quick help.

aivarannamaa, 2023-01-28:
@Patrykchylinski, please reopen if you have answers to my questions above.

closed 2023-01-28 by aivarannamaa