thonny/thonny #2596: PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialTimeoutException:

MatsPojken, 2023-01-02:
After trying different .uf2 files and performing some ok Micropython coding, I get this error message:

"PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialTimeoutException: Write timeout).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 0."

Stop/Restart or changeing pico board or remove followed by reinstall Thonny wouldn't help. The error is still there.

MatsPojken, 2023-01-02:
I dragged my backend.log to here and here is the link:

[backend.log](https://github.com/thonny/thonny/files/10332429/backend.log)

aivarannamaa, 2023-01-04:
Thank you for the report!

Which version of MicroPython are you using? Can you get the REPL on your Picosystem using a simpler tool like `screen` or `tio`?

See similar discussion at https://groups.google.com/g/thonny/c/K7mJ8HIZxK4 and at https://github.com/thonny/thonny/issues/2520#issuecomment-1320976385 (@breaghva)

breaghva, 2023-01-04:
Sorry, but I’m travelling now for some weeks. Will retest this and respond when I returnThanks for following up!Tony

See similar discussion at https://groups.google.com/g/thonny/c/K7mJ8HIZxK4 and at #2520 (comment) ***@***.***)

MatsPojken, 2023-01-04:
I am using MicroPython v1.19.1.
I managed to get rid of the problem by reinstalling MicroPython 1.19.1 in the pico and chosing "Try to detect port automatically" !!
Thanks for the engagement! Please tell the other "poors" about this.
We'll meet again - for sure.

MatsPojken, 2023-01-04:
Thanks for the engagement! Please tell the other "poors" about this.
We'll meet again - for sure.

closed 2023-01-04 by MatsPojken