thonny/thonny #2597: Unable to open Thonny

italiancroydoner, 2023-01-02:
Hello everyone.
Time ago I installed Thonny, and for a while it ran smoothly.
Then, one day, out of the blue, I was unable to start it up.
All I got was the annoying message in the attached file.
I have uninstalled and re-installed it countless times, but to no avail: the same [error 13 Permission denied] message keeps popping up.
For a while I thought that there was some interference from other programs, such as IDLE, but the same thing kept going on whether IDLE was installed or not.
Can someone please help?

Kindest regards

Giorgio

[image]

aivarannamaa, 2023-01-04:
Thank you for the report!

Please check the file mentioned in the last line of the error message. You need to enable showing hidden files in order to see the `AppData` folder. If possible, then delete the frontend.log and try launching Thonny again.

I don't know what may have happened to this file. Did you try restarting your system?

aivarannamaa, 2023-01-28:
Please reopen if you still have this problem and have more information to add.

closed 2023-01-28 by aivarannamaa