thonny/thonny #2601: Unable to save files to device (PICO)

wasatchtechnology, 2023-01-08:
I'm connecting to a raspberry pi PICO. I've used Thonny for weeks successfully untill today. I have been unable lately to save py files to the device. I am running micropython fresh install today. rp2-pico-w-20221220-unstable-v1.19.1-782-g699477d12.uf2.

 I get the following error:
Could not write next block after having written 0 bytes to /sensor_loop_with_web.py
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (OSError: [Errno Could not complete file writing] __W(b'from machine import Pin, WDT\nimport utime, time\nimport network, urequests\n\nwatch_dog = WDT(timeout = 8000)\nmin_dist = 5.0\ntarget_dist = 6.0\nmax_dist = 7.0\nlast_distance = 0\n\nbeeper = Pin(0, Pin.OUT) # GP0\ntrigger = Pin(22, Pin.OUT) # GP22\necho = Pin(7, Pin.IN) # GP 7\nled_target = Pin(3, Pin.OUT) # GP3\nled_too_far = Pin(4, Pin.OUT)# GP4\nled_too_close = Pin(5, Pin.OUT) # GP5\ntarget_set_button = Pin(6, Pin.IN, Pin.PULL_UP) # GP6\nonboard_led = Pin("LED", Pin.OUT) # onboard LED\nfileName = "dim_settings.txt"\non_target_beeps = 0\n\n\ndef measure_distance():\n    global last_distance\n    onboard_led.value(1)\n    \n    trigger.low()\n    utime.sleep_us(2)\n    trigger.high()\n    utime.sleep_us(5)\n    trigger.low()\n   \n    while echo.value() == 0:\n        signaloff = utime.ticks_us()\n    while echo.value() == 1:\n        signalon = utime.ticks_us()\n    timepassed = signalon - signaloff\n    distance = (timepassed * 0.0343) / 2\n    print("The distance from object is ",distance,"cm")\n    \n    distanceInInches = round(distance / 2'): '').
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.

**The BACKEND.log shows the following:**
12:35:57.779 INFO    thonny: Thonny version: 4.0.1
12:35:57.779 INFO    thonny: cwd: /
12:35:57.779 INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem14343401', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:35:57.779 INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
12:35:57.779 INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem14343401', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:35:57.779 INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
12:35:57.779 INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0)
12:35:57.779 INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/cu.usbmodem14343401', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
12:35:57.784 INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 64, write_block_delay: 0, read_block_size: 0
12:35:57.784 INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type RP2040Backend
12:35:57.784 INFO    thonny: TIME/MODS 0.077 before prepare (+73 modules)
12:35:57.784 INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
12:35:57.785 INFO    thonny.plugins.micropython.mp_back: Sending interrupt
12:35:57.785 INFO    thonny.plugins.micropython.mp_back: Done sending interrupt
12:35:57.836 INFO    thonny.plugins.micropython.bare_metal_backend: Poking
12:35:57.875 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'OK\r\n'
12:35:57.875 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: 'MPY: soft reboot\r\n'
12:35:57.890 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
12:35:57.903 INFO    thonny: TIME/MODS 0.119 got welcome 
12:35:57.903 INFO    thonny: TIME/MODS 0.000 bef preparing helpers 
12:35:57.903 INFO    thonny.plugins.micropython.mp_back: Preparing helpers
12:35:57.904 INFO    thonny: TIME/MODS 0.001 befsubcode 
12:35:57.905 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: '\r\n'
12:35:57.916 INFO    thonny.plugins.micropython.bare_metal_backend: Discarding stdout: ''
12:35:57.954 INFO    thonny: TIME/MODS 0.050 affsubcode 
12:35:58.034 INFO    thonny: TIME/MODS 0.080 affforw 
12:35:58.034 INFO    thonny: TIME/MODS 0.000 prepared helpers 
12:35:58.034 INFO    thonny: TIME/MODS 0.000 befsubcode 
12:35:58.036 INFO    thonny: TIME/MODS 0.002 affsubcode 
12:35:58.051 INFO    thonny: TIME/MODS 0.015 affforw 
12:35:58.051 INFO    thonny: TIME/MODS 0.000 got cwd 
12:35:58.051 INFO    thonny: TIME/MODS 0.000 befsubcode 
12:35:58.054 INFO    thonny: TIME/MODS 0.002 affsubcode 
12:35:58.067 INFO    thonny: TIME/MODS 0.013 affforw 
12:35:58.067 INFO    thonny: TIME/MODS 0.000 prepared 
12:35:58.067 INFO    thonny.plugins.micropython.mp_back: Prepared
12:35:58.067 INFO    thonny: TIME/MODS 0.000 befsubcode 
12:35:58.069 INFO    thonny: TIME/MODS 0.002 affsubcode 
12:35:58.092 INFO    thonny: TIME/MODS 0.022 affforw 
12:35:58.092 INFO    thonny: TIME/MODS 0.000 befsubcode 
12:35:58.099 INFO    thonny: TIME/MODS 0.007 affsubcode 
12:35:58.135 INFO    thonny: TIME/MODS 0.036 affforw 
12:35:58.135 INFO    thonny: TIME/MODS 0.000 befsubcode 
12:35:58.146 INFO    thonny: TIME/MODS 0.011 affsubcode 
12:35:58.181 INFO    thonny: TIME/MODS 0.035 affforw 
12:35:58.181 INFO    thonny.backend: Loading Jedi
12:35:58.181 INFO    thonny: TIME/MODS 0.000 Before loading Jedi 
[... cut ...] ce():\n    global last_distance\n    onboard_led.value(1)\n    \n    trigger.low()\n    utime.sleep_us(2)\n    trigger.high()\n    utime.sleep_us(5)\n    trigger.low()\n   \n    while echo.value() == 0:\n        signaloff = utime.ticks_us()\n    while echo.value() == 1:\n        signalon = utime.ticks_us()\n    timepassed = signalon - signaloff\n    distance = (timepassed * 0.0343) / 2\n    print("The distance from object is ",distance,"cm")\n    \n    distanceInInches = round(distance / 2'): ''

aivarannamaa, 2023-01-09:
Thank you for the report!

Do you have enough free space on your flash?

wasatchtechnology, 2023-01-09:
That was an existing python program I was editing when the problem started. I tried so “save as” some very small coupkeline programs and they wouldn’t copy over either. So I don’t believe storage space is the problem.

wasatchtechnology, 2023-01-09:
I found the problem. At some point the requests lib got copied to the PICO and that was the item that used most of the storage space in the device. My learning curve had a speed bump in it.

closed 2023-01-09 by wasatchtechnology