thonny/thonny #2602: Modifying built-in keystroke

Segilmez06, 2023-01-08:
I want to change the <kbd>Ctrl</kbd>+<kbd>K</kbd> keystroke's command to comment out the selection but it's already bound to another one. I've looked but I couldn't find it in the code. Where can I find the line that binds this keystroke? (Or how can I unbind it?)

Segilmez06, 2023-01-09:
I added this lines:
```py
    self.bind("<Control-k>", self._redirect_ctrlk, True)
```
```py
    def _redirect_ctrlk(self, event):
        # Commenting needs to be invoked here
        return "break"
```
Now it disables the built-in keystroke but also doesn't trigger my custom-defined action. How can I invoke the toggle comment function here?

aivarannamaa, 2023-01-09:
Follow the pattern I used for ctrld -- generate a virtual event (e.g. `<<CtrlKInText>>`) and add binding to in commenting plugin https://github.com/thonny/thonny/blob/b857046a563dfae50d46a6fa8ff37546cc02f67f/thonny/plugins/commenting_indenting.py#L184 similar to https://github.com/thonny/thonny/blob/8b1bb6ecacbd093138e31335b99c11d80ccd3e54/thonny/running.py#L197

Segilmez06, 2023-01-09:
Now it's working flawlessly. Thanks for your help.

closed 2023-01-09 by Segilmez06