thonny/thonny #2603: Pygame and .ttf files left open

trex101, 2023-01-09:
In Pygame, I'm wondering if there is a way to close .ttf files after they've been used in TEXT or BUTTON statements. I experienced program halts and "too many files open" when these statements were in loops and although I managed to remove them from the loops and the program is now running smoothly, it still uses repeated calls to update text and buttons so I'd like to ensure the .ttf files and their handlers are closed. 

Typical BUTTON statement:

def get_font(size): 
    return pygame.font.SysFont("notosans", size)

def download():
        DOWNLOAD_BACK = Button(image=None, pos=(1100, 680), 
                            text_input="BACK", font=get_font(20), base_color= gray, hovering_color=white)
        DOWNLOAD_BACK.changeColor(DOWNLOAD_MOUSE_POS)
        DOWNLOAD_BACK.update(SCREEN)

aivarannamaa, 2023-01-09:
Is this issue specific to Thonny? If not, then please ask it in a Pygame forum.

trex101, 2023-01-10:
Understood and done.

closed 2023-01-10 by aivarannamaa