thonny/thonny #2606: Thonny  update

matehubert, 2023-01-10:
Hello,

Yesterday I updated Thonny for the latest version, today it doesn't work:

Python 3.10.6 (C:\Program Files (x86)\Thonny\python.exe)
>>> PROBLEM IN THONNY'S BACK-END: mainloop error (AttributeError: partially initialized module 'socket' has no attribute 'AF_INET' (most likely due to a circular import)).
See Thonny's backend.log for more info.

Process ended with exit code 17.

I tried to reinstall it, the same error, re installed python, same error, if i change interpretter got other errors.

What should I do?

Thanky,

aivarannamaa, 2023-01-10:
Thank you for the report!

Could you please send me the backend.log? You can find it by selecting "Tools => Open Thonny data folder".

matehubert, 2023-01-10:
Sure here is it!
[backend.log](https://github.com/thonny/thonny/files/10382264/backend.log)

aivarannamaa, 2023-01-10:
> Yesterday I updated Thonny for the latest version

You mean Thonny 4.0.1, right?

The error happens right after opening Thonny, right?

Have you installed any extra modules?

aivarannamaa, 2023-01-10:
I now noticed a thing in your log:

> AttributeError: module 'os' has no attribute 'uname'

Do you happen to have a directory named `os` or file named `os.py` in your working directory? If yes, then this will shadow a system module with the same name.

matehubert, 2023-01-11:
I looked after the whole C:// but I doesn't have any files or directories with that name. :/

aivarannamaa, 2023-01-11:
Oh, I didn't pay enough attention, it's the C:\Users\mates\Documents\Python\socket.py, which is shadowing a system module. Please rename this.

aivarannamaa, 2023-01-11:
I'll try to offer some automatic advice in these cases under #384.

matehubert, 2023-01-11:
Thank you for your fast reply it works now. But this file was almost 2 years old and I don't have any problem with it with older versions.

Thank you once again!

closed 2023-01-11 by aivarannamaa