thonny/thonny #2608: Thonny not compatible with new computer?

medellia44, 2023-01-10:
Hi! My daughter does coding with thonny and I am trying to help her.  We just got a new Mac running the Ventura OS.  She is not able to open the thonny program or the Python Launcher. I tried deleting and reinstalling as well.  When we click the icon a mostly blank window pops up with the Apple settings icon (gray box w/gear).  Is thonny not compatible with this OS or are we doing something wrong??

aivarannamaa, 2023-01-11:
Thank you for the report!

Is this Arm Mac? Which Thonny version did you install?

It could be helpful if you sent me the contents of `~/Library/Thonny` directory (the configuration file and log files)

medellia44, 2023-01-11:
Hi!  It is a desktop Mac if that’s what you mean…I am not sure where to find the log files but if you could walk me through that I’d appreciate it!

aivarannamaa, 2023-01-11:
Does your Mac have Intel (also called x86) processor or Arm processor (eg. M1 or M2)?

The log files are located in the `Library/Thonny` directory inside your home directory.

medellia44, 2023-01-11:
It has an M1 chip.  Is there a tutorial on how to find files in the home directory?  I am not that computer literate but trying to help out my young daughter!  Thanks!

aivarannamaa, 2023-01-12:
> Is there a tutorial on how to find files in the home directory?

You need to open Finder. It should show your home folder at first. From there you should find a folder named "Library" and inside this there should be folder named "Thonny" and inside this there should be files configuration.ini, backend.log and frontend.log.

It would be great if you could post these files here.

medellia44, 2023-01-12:
OK got it!  I tried adding the configuration.ini but it says that file type isn't supported...

[backend.log](https://github.com/thonny/thonny/files/10401357/backend.log)
[frontend.log](https://github.com/thonny/thonny/files/10401358/frontend.log)

aivarannamaa, 2023-01-12:
Thanks!

Which Thonny version did you install? These log files are created with an older Thonny version (3.3.13), but I'd recommend trying the latest version (4.0.1).

medellia44, 2023-01-12:
I did go to the thonny.org website multiple times and downloaded/installed the newest version, but the log files keep showing the 3.13 version.  Not sure if there's a step I'm missing.

aivarannamaa, 2023-01-13:
Can you check the modification time of the log files? Maybe the new version doesn't get to write the log file.

medellia44, 2023-01-13:
Yes, the log files are showing a date of Dec 24.

aivarannamaa, 2023-01-13:
> Yes, the log files are showing a date of Dec 24.

This means that on Dec 24 you have ran Thonny 3.3.13 on this machine. Was this successful?

medellia44, 2023-01-13:
It ran on our old computer.  We transferred everything to an external hard drive and then moved it to our new computer and it has not worked since!

aivarannamaa, 2023-01-13:
> We transferred everything to an external hard drive and then moved it to our new computer and it has not worked since!

I see. In general this is not recommended approach for programs. It's best to re-install programs from scratch, especially if you move from Intel processor to M1 processor.

If you later re-installed, then you may now have two Thonny versions on your computer. Try deleting all "Thonny"-s you find in your computer (on Desktop, under Applications), then delete `Library/Thonny` and the re-install Thonny 4.0.1.

medellia44, 2023-01-13:
I tried that but I am still getting the same error.  And when I look in the Library, there is now no folder at all for Thonny.

aivarannamaa, 2023-01-14:
In this case I have no more ideas what to check. If you can, then you could send me a screencast, which demonstates how do you launch Thonny and what happens after this. Maybe this gives some hints.

medellia44, 2023-01-14:
[image]

medellia44, 2023-01-14:
That window on the left is what happens when I try to open Thonny.  I wonder if I should reach out to Apple about this - by the way, thank you SO much for all your help!

aivarannamaa, 2023-01-16:
> I wonder if I should reach out to Apple about this

I don't think they would consider this their responsibility.

Yesterday I released a new version in Thonny which contains updated version of Python. There's a chance that it may work, so you could try installing https://github.com/thonny/thonny/releases/download/v4.0.2/thonny-4.0.2.pkg

I have a question about your screenshot. I don't see Thonny icon anywhere -- can you please describe how did you launch Thonny? By opening Finder => Applications and then double-clicking on Thonny icon? And then you hide the Finder window?

medellia44, 2023-01-16:
I can see the Thonny icon in the Launchpad and that's how I opened it, but it won't show in the dock or anything like that.  When I try clicking it, all i can see is that tiny window in the other screenshot and the only option is to click "done" and close it.

aivarannamaa, 2023-01-17:
Oh, I totally forgot about Launchpad.

Did you try 4.0.2? No improvement with this?

medellia44, 2023-01-18:
Yes I tried the new one you linked...no luck.  It says installation complete and I can see it in launchpad, but it won't actually open...

aivarannamaa, 2023-01-18:
There is one more thing to try... When you launch a program and macOS detects some problems, it writes something into _system log_.

Here are some instructions about opening and browsing macOS system log: https://www.howtogeek.com/356942/how-to-view-the-system-log-on-a-mac/

The procedure would be to open the system log and see whether something interesting turns up when you try to launch Thonny.

mdrescher, 2023-03-14:
I can confirm that I have Thonny 4.0.2 running on mac OS Ventura on my Macbook Air.

Installing Thonny via `pip3 install thonny` was straightforward, but I was not able to use the PyPI integration due to certificate errors.

I then downloaded the installer from thonny.org which includes its own Python bundling, installed it, and all was good, no issues.

nicosuave, 2023-03-22:
Hey @aivarannamaa I am seeing crashing when switching the interpreter to MicroPython / RPi Pico. 

[Here is the contents of `~/Library/Thonny` and the Problem Report after the crash](https://gist.github.com/nicosuave/2901a9563139a5955d9a12eabb9c0d8b)

aivarannamaa, 2023-03-26:
@nicosuave, it looks like your case is quite different from the case investigated in this thread. If your problem persists and you are using latest Thonny version then please open a separate issue.

kittytg, 2023-11-20:
we are having same issue with MacOS with Thonny
OS is Ventura 13.5 while thonny is the latest one.
we tried to uninstall it many times
[backend.log](https://github.com/thonny/thonny/files/13417914/backend.log)
[frontend_faults.log](https://github.com/thonny/thonny/files/13417915/frontend_faults.log)
[frontend.log](https://github.com/thonny/thonny/files/13417916/frontend.log)
still error said cant located python

aivarannamaa, 2023-11-20:
@kittytg, thanks for the logs! Unfortunately they don't say anything about not being able to locate Python. Can you make a screencast of what is happening?

vsquared, 2024-05-19:
I'm unable to run test code using Thonny with a py5 plugin on a new Mac M2.  An identical error message was thrown by another new M2 owned by a friend.  The error message is shown below:
```
    import py5
  File "/Users/xxxx/Library/Python/3.10/lib/python/site-packages/py5/__init__.py", line 75, in <module>
    raise RuntimeError("py5 is unable to start Java 17 Virtual Machine")
RuntimeError: py5 is unable to start Java 17 Virtual Machine
```
Runs ok on a mac with the old Intel chip, but this appears to be an issue with the newer M2 chips.