thonny/thonny #2609: Thonny 4.0.1 can not remote python(SSH)  to raspberry pi

sakunamary, 2023-01-11:
HI guys :
  I am trying to make Thonny 4.0.1 connect to remote raspberry pi  .BUT it dos't work .Shell show following msg 
  FYI: MacBook with X86 chips and MacOS 13.0.1 ,SSH login  raspberry pi  is OK .python3.9.2 on raspberry pi .
I have install paramiko in Thonny too.
Thanks for your help!

Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/paramiko/sftp_client.py", line 130, in __init__
    server_version = self._send_version()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/paramiko/sftp.py", line 134, in _send_version
    t, data = self._read_packet()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/paramiko/sftp.py", line 201, in _read_packet
    x = self._read_all(4)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/paramiko/sftp.py", line 188, in _read_all
    raise EOFError()
EOFError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 198, in <module>
    backend = SshCPythonBackend(**args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 35, in __init__
    self._upload_main_backend()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 160, in _upload_main_backend
    if self._get_stat_mode_for_upload(launch_dir) and not thonny.get_version().endswith("-dev"):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 880, in _get_stat_mode_for_upload
    return self._perform_sftp_operation_with_retry(lambda sftp: sftp.stat(path).st_mode)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 872, in _perform_sftp_operation_with_retry
    return operation(self._get_sftp(fresh=False))
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/backend.py", line 833, in _get_sftp
    self._sftp = paramiko.SFTPClient.from_transport(self._client.get_transport())
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/paramiko/sftp_client.py", line 170, in from_transport
    return cls(chan)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/paramiko/sftp_client.py", line 132, in __init__
    raise SSHException("EOF during negotiation")
paramiko.ssh_exception.SSHException: EOF during negotiation

Process ended with exit code 1.

aivarannamaa, 2023-01-11:
Thank you for the report!

I'm not able to reproduce with RPi OS Bullseye and Intel Catalina.

Are you using password or SSH key authentication?

sakunamary, 2023-01-11:
> Thank you for the report!
> 
> I'm not able to reproduce with RPi OS Bullseye and Intel Catalina.
> 
> Are you using password or SSH key authentication?
   
I am using password .

Thanks and have a good day!

aivarannamaa, 2023-01-11:
How did you configure ssh server on your Pi? Which PiOS version are you using?

sakunamary, 2023-01-12:
> How did you configure ssh server on your Pi? Which PiOS version are you using?
FYI:
raspi is running this OS.-> http://dietpi.com. which is based on Debian . And ssh server is [Dropbear - Lightweight SSH Server](https://dietpi.com/docs/software/ssh/#dropbear) . 

I can login in as by macOS terminal (zsh)  as  usual .

Thanks!

aivarannamaa, 2023-01-12:
> And ssh server is [Dropbear - Lightweight SSH Server](https://dietpi.com/docs/software/ssh/#dropbear).

That's important. It looks like Dropbear doesn't have the features Paramiko (the SSH library used Thonny) is expecting: https://stackoverflow.com/a/48446801/261181

sakunamary, 2023-01-12:
> > And ssh server is [Dropbear - Lightweight SSH Server](https://dietpi.com/docs/software/ssh/#dropbear).
> 
> That's important. It looks like Dropbear doesn't have the features Paramiko (the SSH library used Thonny) is expecting: https://stackoverflow.com/a/48446801/261181

OH！ That's important in deed :-) .I will switch to raspberry OS later ,and double check !

Thanks you so much !

sakunamary, 2023-01-12:
> > And ssh server is [Dropbear - Lightweight SSH Server](https://dietpi.com/docs/software/ssh/#dropbear).
> 
> That's important. It looks like Dropbear doesn't have the features Paramiko (the SSH library used Thonny) is expecting: https://stackoverflow.com/a/48446801/261181

I have checked. the reason that why Thonny can not ssh to raspberrypi  is Dropbear ...I ssh to another device which is  run Armbian OS .It is working GOOD!

closed 2023-01-12 by sakunamary

lizard20, 2023-06-16:
Could not force-stop the program. Please wait, try again or select Stop/Restart!%Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "/opt/thonny/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 213, in <module>
    backend = SshCPythonBackend(**args)
  File "/opt/thonny/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 45, in __init__
    self._proc = self._start_main_backend()
  File "/opt/thonny/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 153, in _start_main_backend
    return self._create_remote_process(
  File "/opt/thonny/lib/python3.10/site-packages/thonny/backend.py", line 814, in _create_remote_process
    raise RuntimeError(f"Got {ack!r} instead of expected {PROCESS_ACK!r}")
RuntimeError: Got 'This version of Thonny only supports Python 3.8 and later.' instead of expected 'OK'

Process ended with exit code 1.
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Thonny 4.1.1, can't remote python3 ssh to Raspberry Pi 3

$ uname -a
Linux raspberrypi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

---------------------------------------------------------------------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/thonny/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 213, in <module>
    backend = SshCPythonBackend(**args)
  File "/opt/thonny/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 45, in __init__
    self._proc = self._start_main_backend()
  File "/opt/thonny/lib/python3.10/site-packages/thonny/plugins/cpython_ssh/cps_back.py", line 153, in _start_main_backend
    return self._create_remote_process(
  File "/opt/thonny/lib/python3.10/site-packages/thonny/backend.py", line 814, in _create_remote_process
    raise RuntimeError(f"Got {ack!r} instead of expected {PROCESS_ACK!r}")
RuntimeError: Got 'This version of Thonny only supports Python 3.8 and later.' instead of expected 'OK'

Process ended with exit code 1.

aivarannamaa, 2023-07-04:
@lizard20 , your error message seems to say your remote Python version isn't recent enough. I made a new issue for this: #2847